fivdi/pigpio #2: "Sorry, you don't have permission to run this program."

sebastiaanluca, 2015-12-11:
```
2015-12-11 15:36:28 initCheckPermitted: 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+
Error: pigpio error -1 in gpioInitialise
```

Thrown in https://github.com/rpicopter/pigpio/blob/master/pigpio.c#L3697

Running my app as root using sudo throws other exceptions, so is there anything I can do to make the library run as a non-root user?

fivdi, 2015-12-11:
Root privileges are needed to access the hardware required to enable features like reliable PWM on any pin. What exceptions are thrown when running as root?

sebastiaanluca, 2015-12-11:
My scripts depend on being run as a certain user. For instance to access its files. Some other modules also throw generic errors, too many to list.

Can I chmod the files that are root-only so my user can access them? What's their location?

fivdi, 2015-12-12:
The file in question is `/dev/mem`. However changing the permissions of `/dev/mem` will not resolve the issue. What could be used to resolve the issue is `sudo chmod u+s /usr/bin/node`. You'll need to decide for yourself whether or not this is something that you want to do.

sebastiaanluca, 2015-12-12:
Yeah, chmodding `/dev/mem` didn't work. Tried the same for the `/usr/local/include` files and other where the libgpio files are located, but without result. Somehow I did manage to get it running with sudo in the end, but I don't remember how o_O Anyway, glad it's working. Too bad I then figured out I can't read from a potentiometer / rotary control :)

closed 2015-12-12 by sebastiaanluca

Terristen, 2016-01-11:
@sebastiaanluca
I'm no expert, but you can read a resistance-based analog input, like a rotary pot from a digital gpio pin. That is, if you're willing to accept a margin of error in the readings

Here's the concept:
* If you can measure the amount of time it takes to charge a capacitor of a known capacity (1 micro-farad for example) utilizing current of a known resistor (2.2k for example), then you can place your resistive-sensor/input in serial between the resistor and the capacitor and measure the time it takes to discharge the capacitor to ground.
```
+3.3v -----vvvv-----sensor-----gpio----------||---------=
                 2.2k         pot       gpio16        1mf         ground
```
The algorithm is simple:
1. set the pin to output and write(0)
2. set the pin to input and set a variable to register the time
3. wait for the pin to switch to High and measure the time again
4. the difference between time 2 and time 1 is proportional to the amount of resistance in the potentiometer

What #4 means is that, the less resistance the pot is providing, the more current will flow to the capacitor. The more current that flows to the capacitor, the faster it will reach ~63% of capacity (the mathematical point that it will trigger a HIGH interrupt on the GPIO pin). Inversely, the higher the resistance, the slower the capacitor charges. 

So at one end of the pot you might measure 2 milliseconds, and at the other you might measure 50 milliseconds. The time-frame is greatly dependent on the resistor and capacitor you select. If your sensor is highly resistive, you might raise your cap-size, and if it is fairly low resistive, you might make the cap smaller.

The circuit I used to measure a Light-Resistive-Diode included a 2.2k resistor and a 3.3k cap. My milliseconds ranged in the 2ms to 17 ms range (sensor exposed to room lighting to fully occluded). 

I'm still experimenting with the circuit, but I did prove the concept. My current troubles are two-fold: 
* Initial values can be really high, when testing after kicking off the script. This could be circuit warm-up, or attributable to node settling in after initializing the script.
* when testing a stable pot (I was using a 2-axis joystick module) the measured time can fluctuate as much as 3 ms between consecutive measurements, with no changes being applied to the pot itself.

I have some theories as to how to normalize these kinds of readings, but it will never be as accurate as an analog input. (though, as I recall, even those had some jitter on the arduino)

sample code

```javascript
//pardon the use of onoff, I haven't converted this to pigpio 
//yet because I'm having sudo issues like you had
var GPIO = require('onoff').Gpio,
	moment = require('moment');

function readAnalog(pin){
	var thispin = new GPIO(pin,'out');

	thispin.setDirection('out');
	thispin.writeSync(0);
	var timer = moment();
	thispin.setDirection('in');
	while(thispin.readSync() < 1){
		if(moment().diff(timer) >= 2000) {
			return "Read Timeout Exceeded";
		}
	}
	var result = moment().diff(timer);
	
	thispin.setDirection('out');
	thispin.writeSync(0);
	
	return result;
}
```

Terristen, 2016-01-11:
Caveat on the post above, Pi-browser doesn't let me edit my post, so bear with the sloppy electronics terminology. :)

Also, the sample code us reading synchronously, but if your application warranted it, you could use timeouts or intervals, or even interrupts to make the code less blocking.

fivdi, 2016-01-11:
@Terristen The author of the pigpio C library (which is used by the pigpio Node.js package) also describes this technique [here](http://abyz.co.uk/rpi/pigpio/ex_LDR.html).

>Initial values can be really high, when testing after kicking off the script. This could be circuit warm-up, or attributable to node settling in after initializing the script.

Perhaps it's could be related to the loading of Node.js modules or just-it-time compilation.

>when testing a stable pot (I was using a 2-axis joystick module) the measured time can fluctuate as much as 3 ms between consecutive measurements, with no changes being applied to the pot itself.

Given that Linux is a multitasking Operating System, I think it's going to be difficult to measure time accurately using the technique that's currently being used. For example, in the following snippet of code, which is taken from above, the Operating System may decide to reschedule the CPU after the loop has completed but before the time diff has been measured. This will mess up the timing. Perhaps this is why the measured time can fluctuate as much as 3 ms between consecutive measurements, with no changes being applied to the pot itself.

```js
    while(thispin.readSync() < 1){
        if(moment().diff(timer) >= 2000) {
            return "Read Timeout Exceeded";
        }
    }
    // The Operating system may decide to reschedule the CPU to a different process/thread
    // after the loop has completed but before the time has been measured.
    var result = moment().diff(timer);
```

The pigpio C library has a function called [gpioSetAlertFunc](http://abyz.co.uk/rpi/pigpio/cif.html#gpioSetAlertFunc) which functions a little like interrupt routines but provides fairly accurate timing/tick information. This is likely to give better results.

I could be persuaded to provide bindings for gpioSetAlertFunc in the pigpio Node.js module :smile:.

Terristen, 2016-01-11:
@fivdi I agree. The multitasking features of the OS are the most likely culprit there. I plan on implementing my experiment with pigpio as soon as I can get it running... at which point I might be asking for gpioSetAlertFunc. :)

To the point of this original thread, though, I've got the same issue with pigpio as the OP. I'm using Nodeclipse and using the project run configuration "as a nodejs application." I'm fairly new to Linux, so my understanding of the vagaries and magic behind the scenes with permissions is still pretty basic.

A colleague (more nuanced in his understanding of Linux) suggested I `chown 777` the core PIGGPIO folder. Another recommended starting pigpio daemon as root on startup. Your thoughts?

fivdi, 2016-01-11:
The pigpio Node.js package uses the pigpio C library and the pigpio C library requires root privileges, so root privileges are needed. The pigpio deamon is not used, so starting it will not resolve the issue. The idea behind using the pigpio C library directly is to make things as fast possible. Typically, the node executable is owned by root and in /usr/bin:
```
pi@raspberrypi:/usr/bin $ ls -la node
lrwxrwxrwx 1 root root 6 Apr 12  2015 node -> nodejs
```
The [setuid](https://en.wikipedia.org/wiki/Setuid) access right could be used to allow user pi to run node with the rights of root with the following command, but this is a crazy thing to do:
```
sudo chmod u+s /usr/bin/node
```

stefanomasini, 2016-01-17:
Thanks for the great bindings!
I understand the need for maximum speed, but I'm using this library together with google coder, where it makes a lot of sense to not have node running as root.
Any plans to optionally support using the separate daemon, maybe to the expense of a bit of speed?

fivdi, 2016-01-18:
No plans at the moment.

dkebler, 2017-05-18:
@fivdi probably worth pointing this out in the readme install section along with your comment about using chmod and cavet above.  Many wouldn't think to run their script as root as that is uncommon.

----
Here is my solution which is a bit more secure but still requires using sudo but will work for headless where entering a sudo password is not an option.

Have a sudoer account for running code  in my case `sysadmin` that can ONLY be entered via ssh with with a keypair.

Now
Add this a file, say `gpio-node`, in /etc/sudoers.d/
```
# group gpio can also run node for headless access to pigpio library
%gpio ALL=NOPASSWD: /usr/local/bin/node
```
make sure (add) `sysadmin` is in the `gpio` group
note:  don't use ` /usr/bin/node` as NOPASSWD doesn't work with links! and that's a link not `node` itself.  Any issue make sure that /usr/local/bin is before  /usr/bin in your path.  BTW that "node" links to nodejs in the same directory that is hard linked to node in /usr/local/bin.

now in your package.json write any scripts which will call pigpio library with `sudo node`  instead of `node`.   You shouldn't be asked for a  password.

As extra security since I only have a single user on my headless machine I  chowned the group to `node` and removed running node by everyone.   
```
sudo chmod 754 /usr/local/bin/node
sudo chown root:sysadmin /usr/local/bin/node

-rwxr-xr-- 1 root sysadmin 23888496 May  3 15:39 node*
```
of note you can still run node without sudo fine albiet with the last suggestion only from `sysadmin`