fivdi/pigpio #3: Bit Banging Operations

JoelDapello, 2016-03-03:
Hello! Awesome work with this library. I'm wondering if you have plans or interest in supporting the bit banging operations capable with the pigpio c library? 

Specifically, I'm interested in the TX associated functions
```
gpioWaveClear
gpioWaveAddNew
gpioWaveAddGeneric
gpioWaveAddSerial	
gpioWaveCreate	
gpioWaveDelete	
gpioWaveTxSend
gpioWaveTxBusy
```
and the RX associated functions
```
gpioSerialReadOpen
gpioSerialRead
gpioSerialReadClose
```

If you are not planning to add these, perhaps I could assist in integrating them?

fivdi, 2016-03-03:
I'm interested in supporting as much of the pigpio C library as possible :). If adding support for the bit banging operations is something that you would like to do then feel free to do so.

fivdi, 2017-06-10:
I've decided against adding the bit banging operations.

closed 2017-06-10 by fivdi