fivdi/pigpio #4: configureClock() doesn't seem to work

alberto2000, 2016-05-11:
I can't get configureClock(microseconds, peripheral) to do anything - probably it's just me doing it wrong so I'm sorry this might not be an issue but more of a question - how to use configureClock() so to set resolution to 2 instead of default 5?

alberto2000, 2016-05-11:
I found it out - had to run require('pigpio').configureClock(2, 0); prior to creating the main Gpio object - Closing.

fivdi, 2016-05-12:
Yes, configureClock needs to be called before creating objects. This should be documented.

alberto2000, 2016-05-12:
Ok. Isn't this also a bit of a strange programming pattern in the sense that the configuration being set while actually creating the object would be more streamlined?

fivdi, 2016-05-12:
It probably looks strange because it's not documented very well at the moment. The clock being configured corresponds to a hardware peripheral on the Raspberry Pi; either the PWM peripheral or PCM peripheral. It's a global clock that's used for multiple purposes simultaneously, for example, PWM and alerts.

The same clock is used by all objects. For this reason it doesn't make sense to configure the clock when creating an object.

The pigpio C library which is used by the pigpio JavaScript add-on requires that the clock be configures before using any other functions (more or less.)

fivdi, 2016-05-12:
Code to call configureClock could look like this:
```js
var pigpio = require('../'),
  Gpio = pigpio.Gpio,
  led;

pigpio.configureClock(1, pigpio.CLOCK_PCM);

led = new Gpio(18, {
  mode: Gpio.OUTPUT,
  alert: true
});
...
```

fivdi, 2016-06-04:
The documentation for `configureClock` has been improved and hopefully things are a lot clearer now. See https://github.com/fivdi/pigpio/blob/master/doc/configuration.md#configureclockmicroseconds-peripheral

closed 2016-06-04 by fivdi

alberto2000, 2016-06-08:
@fivdi thanks for this