fivdi/pigpio #6: Signal Handler

karthikbnd, 2016-06-06:
Hai,
When i execute the program in loop and terminates it by pressing CTRL+C it gives me this signal handler

 sigHandler: Unhandled signal 2, terminating 

How to over come this

fivdi, 2016-06-06:
The pigpio Node.js package is a wrapper for the pigpio C library. The pigpio C library will terminate successfully on all unhandled signals. For example, if ctrl+c is hit and not handled, the pigpio C library will terminate successfully. However, because the signal was not handled, the pigpio C library will also print the message `"sigHandler: Unhandled signal 2, terminating"`. This is the intended behavior and doesn't mean that there is an issue.

To prevent this message from being printed, a `SIGINT` handler can be implemented. From the [GPIO Cleanup](https://github.com/fivdi/pigpio/issues/5) issue that you created earlier today, what I think you want to do is hit ctrl+c to terminate your program, but before the program terminates, you would like to clear the value of some GPIOs. This could be achieved with code similar to that shown below. The program toggles the state of an LED connected to GPIO17 once per second. If ctrl+c is hit, the `SIGINT` handler is called. The handler turns the LED off and clears the timer interval `iv` so that the program can terminate gracefully. The message `"sigHandler: Unhandled signal 2, terminating"` will not be printed to the screen because ctrl+c (`SIGINT`) is being handled.

```js
var Gpio = require('pigpio').Gpio,
  led = new Gpio(17, {mode: Gpio.OUTPUT}),
  iv;

var iv = setInterval(function () {
  led.digitalWrite(led.digitalRead() ^ 1);
}, 1000);

process.on('SIGINT', function () {
  led.digitalWrite(0);
  clearInterval(iv);
});
```

fivdi, 2016-06-07:
@karthikbnd did the above response answer the question?

karthikbnd, 2016-06-08:
yes
it works perfect
thanks

closed 2016-06-08 by karthikbnd

Rmannn, 2016-08-15:
This is not working when using the servo
```js
const Gpio = require('pigpio').Gpio;
const serviceButton = new Gpio(18, {
    mode: Gpio.INPUT,
    pullUpDown: Gpio.PUD_OFF,
    edge: Gpio.RISING_EDGE
});

[...]

process.on('SIGINT', ()=> {
    serviceButton.digitalWrite(0);
});
```

This results in 

> sigHandler: Unhandled signal 2, terminating

javadoug, 2016-09-05:
@fivdi I'm noticing similar behavior when I restart my pigpio app using systemctl. Sometimes I get sigHandler: Unhandled signal 18, terminating and it hangs until the sigterm timeout; sometimes I get sigHandler message with different signal numbers 2 and 11, I think. And sometimes I get memalloc errors. What is the best way to clean up? 

My app was originally implemented with pigpiod daemon and it worked without these issues. So I'm wondering if there is something in the binding coder or, it wouldn't surprise me, if there is something I'm missing.

```
//http://joseoncode.com/2014/07/21/graceful-shutdown-in-node-dot-js/
function shutdown() {
	gpioPins.unexport(); // call disableInterrupt and disableAlert
	server.close(function () { // close express/http server
	  	setTimeout(function () {
			process.exit(0);
		}, 300);
	});
}

process.on('SIGINT', shutdown);
process.on('SIGTERM', shutdown);
```

javadoug, 2016-09-05:
OK I changed a couple things and now I get a clean shutdown.

1. moved the exit out of the close callback. close did not call callback bc no new requests come in and so close just waits for connection to timeout and thus process.exit(0) never got called.
2. added signal handler listeners for SIGHUP and SIGCONT
3. extended the timeout before calling process.exit(0) to give app more time to clean up

```
//http://joseoncode.com/2014/07/21/graceful-shutdown-in-node-dot-js/
//https://github.com/nodejs/node-v0.x-archive/issues/9066
function shutdown() {
	gpioPins.unexport();
	server.close();
	setTimeout(function () {
		process.exit(0);
	}, 3000);
}

process.on('SIGHUP', shutdown);
process.on('SIGINT', shutdown);
process.on('SIGTERM', shutdown);
process.on('SIGCONT', shutdown);
```