fivdi/pigpio #7: free port when application finishes

javadoug, 2016-08-27:
Problem: pigpio binds on port 8888. when app shuts down the port remains bound. Subsequently, when the app is run again it fails with following error message.

```
 initInitialise: bind to port 8888 failed (Address already in use)
/home/pi/verifiedcut/webapp-server/node_modules/pigpio/pigpio.js:11
    pigpio.gpioInitialise();
           ^

Error: pigpio error -1 in gpioInitialise
    at Error (native)
    at initializePigpio (/home/pi/verifiedcut/webapp-server/node_modules/pigpio/pigpio.js:11:12)
    at new Gpio (/home/pi/verifiedcut/webapp-server/node_modules/pigpio/pigpio.js:25:3)
    at Object.<anonymous> (/home/pi/verifiedcut/webapp-server/server.js:254:29)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.runMain (module.js:590:10)
```

I expected to see/use the C pigpio gpioTerminate method. Is this method exposed in the binding? How do I properly terminate app?

fivdi, 2016-08-28:
I can't reproduce this error. For example, [distance-hc-sr04.js](https://github.com/fivdi/pigpio/blob/master/example/distance-hc-sr04.js) can be run multiple times successfully. Each time it's run, it's terminated with `ctrl-c` and then run again.

To the best of my knowledge, `gpioTerminate` is called automatically by the pigpio C library when the node process terminates (see [here](https://github.com/joan2937/pigpio/blob/master/pigpio.c#L5277), [here](https://github.com/joan2937/pigpio/blob/master/pigpio.c#L5287), and [here](https://github.com/joan2937/pigpio/blob/master/pigpio.c#L7748).)

Can you post a simple program that can be used to reproduce the error?

javadoug, 2016-09-02:
@fivdi Yeah, that would suggest an issue with my code. Closing as this is not likely an issue for this codebase.

closed 2016-09-02 by javadoug

fivdi, 2016-09-02:
I opened https://github.com/joan2937/pigpio/issues/81 to verify that my understanding of gpioTerminate is correct.

fivdi, 2016-09-02:
@javadoug Not that it's no possible to use pigpiod and the pigpio JavaScript module at the same time. Was this by any chance the case?

javadoug, 2016-09-02:
No I explicitly checked that. I'm using pigpio in an express app and I did notice strangeness about when express is configured and in relation to configuring the pigpio pins. I'm still reducing my code to the minimum needed to fail. Should know more tomorrow. Will let you know what I learn.

javadoug, 2016-09-03:
OK after removing code until I found the culprit - I can confirm the issue is most likely *not* pigpio. The app runs some ssh commands in a shell script via child_process.execFileSync to validate the code base version before allowing the app to continue with start up tasks. With the ssh commands in the script when the app is shutdown via ctrl-c or systemctl the port 8888 hangs and prevents a subsequent start up as described in this issue. 

If I comment out the ssh commands in the script the app behaves as expected and can start and stop without hanging port 8888. Not sure why the ssh commands would affect the behavior of the pigpio initialize and terminate routines but at least I can confidently close this issue here.

dkebler, 2018-02-22:
Hmm....having the same trouble.  Didn't have this before on a pi3.  Using and rpi, rebooted to make sure something wasn't using that port.
Actually why is pigio needing a tcp port.  Why doesn't it use a named pipe/unix socket if it needs to communicate interprocess??  Can one change the default port??

dkebler, 2018-02-23:
Yea weird this is dietpi install so no extra junk and and there are no tcp servers I started at this port and yet it shows port 8888 in use after I reboot so it can;t be left over unterminated pigpio.   But still why in the world would pigpio need a TCP port for inter or intra process communication and why would it insist on such a commonly used dev port as 8888 anyway.  Makes no sense.  
I'll have to track down where this port use is coming from but if pigpio really does need a tcp port how to I set it away from the default.  I see nothing in the node api
```

Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State      
tcp        0      0 0.0.0.0:ssh             0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:8888            0.0.0.0:*               LISTEN     
tcp6       0      0 [::]:ssh                [::]:*                  LISTEN     
udp        0      0 0.0.0.0:bootpc          0.0.0.0:*                          
Active UNIX domain sockets (only servers)
```

dkebler, 2018-02-23:
ok I tracked it down it was some dietpi system health web monitor.
Still I would like to know why pigpio need to use a TCP port.   I assume that is something in the pigpio C code and not your bindings?  Can it be specified  (can't think of software with hard coded, unsetable TCP ports).  Why since it is for sure going to be run on linux would one not use a linux socket/named pipe.  This is very standard for IPC on the same machine and in general is faster.

dkebler, 2018-02-23:
ok nevermind.....I found #22  so apparently I can switch the port.
```
// Call configureSocketPort before creating Gpio objects
pigpio.configureSocketPort(8889);
```
Still shaking my head for code that needs a TCP port when a named/unix socket would have been the logical choice.

fivdi, 2018-02-24:
> Still shaking my head for code that needs a TCP port when a named/unix socket would have been the logical choice.

If I'm not mistaken the pigpio C library does this to enable one Pi to control another remotely.