fivdi/pigpio #8: interrupt triggers on both edges when leading edge is configured

javadoug, 2016-08-27:
problem: a pin configure with {edge: Pgpio.LEADING_EDGE} still fires falling edge interrupts. Variations on the following code taken from documentation did not work.

```
const Gpio = require('pigpio').Gpio;
const testPin = new Gpio(26, {mode: Gpio.INPUT, pullUpDown: Gpio.PUD_DOWN, edge: Gpio.RISING_EDGE});
testPin.disableInterrupt(); # tried disabling and reenabling to no avail
console.log('interrupt value is', Gpio.RISING_EDGE); # do I actually set a value
testPin.enableInterrupt(Gpio.RISING_EDGE);
testPin.on('interrupt', function (level) {
  console.log(`we got an interrupt ${level}`); #get both edges no matter what
  setTimeout(() => {console.log('ok, after 3 seconds', testPin.digitalRead())}, 3000); # sanity check pin
});

Outputs:
interrupt value is 0
# Set Pin HIGH for ~ 1 second 
we got an interrupt 1
we got an interrupt 0
ok, after 3 seconds 0
ok, after 3 seconds 0
```

This problem existed for me in the C lib as well so may not be a bug in the binding code but if you tested this then perhaps I am doing something wrong which I'd love to discover so I can fix it.

javadoug, 2016-08-27:
the workaround in case someone else has this issue is to return out of the function on the edge you don't want.

fivdi, 2016-08-28:
When using interrupts, the `level` argument passed to the `interrupt` event handler is not the level of the pin the instant the interrupt occurs. It's the level of the pin "shortly after" the interrupt occurs. Linux is not a realtime operating system, so the amount of time that passes between the detection of the interrupt and determining the level of the corresponding pin isn't defined and can be several microseconds later. The level of the pin can change during this time span. Such effects can be seen with buttons or switches that bounce.

Are you using a button or switch that's bouncing?

javadoug, 2016-09-02:
@fivdi I'm counting a rotary encoder A and B channels. The issue is not reading the value after the interrupt, the issue is getting a HIGH and LOW interrupt when I only expect a HIGH interrupt. Unless of course I'm not understanding how the 'edge' option works.

fivdi, 2016-09-02:
> The issue is not reading the value after the interrupt, the issue is getting a HIGH and LOW interrupt when I only expect a HIGH interrupt. Unless of course I'm not understanding how the 'edge' option works.

I think there's a misunderstanding here. The level argument passed to the interrupt event handler is not HIGH for a rising edge interrupt and LOW for a falling edge interrupt. The level argument is the 0 if the pin is 0V and 1 if the pin is 3.3V when the pin level is read by the pigpio JavaScript package after the interrupt occurs. Although the test code posted above only enables rising edge interrupts, the level argument passed to the interrupt event handler can be 0 or 1 if the rotary encoder is bouncing and rotary encoders can bounce.

There's sample C code for a rotary encoder [here](http://abyz.co.uk/rpi/pigpio/ex_rotary_encoder.html) and it should be relatively easy to translate it to JavaScript. The C code uses alerts rather than interrupts. It also handles debouncing.

javadoug, 2016-09-03:
Ah, yes, my understanding on this is probably not right, yet. It's not clear why an interrupt configured for RISING_EDGE would pass a level of 0 to the callback handler. A pin that is HIGH could not possibly rise higher and therefore would not fire a RISING_EDGE interrupt while the pin level is already 1 (and vice versa). I suspect the answer to this might be in the link you pointed me to "Contact A bounces for circa 700 us before completing the level transition". Why wouldn't the interrupt/alert trigger on the "completed level transition" rather than oscillate with the micro changes in the signal? Hmm is the interrupt faster than the read of the gpio level?

Side note, in the example you pointed to I notice the code uses alerts instead of interrupts. What is the difference between interrupts and alerts?

 Anyway, thanks for the help and the pointers.

javadoug, 2016-09-03:
@fivdi if you believe the code is working as designed I'm good with closing this. We can conclude it is my lack of knowledge about how hardware works with software.

closed 2016-09-03 by javadoug