fivdi/pigpio #9: Hardware PWM

shtaif, 2016-09-20:
Hello,

I become more and more confused as PWM seems to me less and less an absolute term.
You declare the word "hardware" a lot in the library, but at the same time, I saw a table that deals with percentages of CPU consumed by varying PWM configurations.

Isn't "Hardware PWM" CPU-independent?
Also, some other library claimed there are only 4 PWM pins on RPI 3 board, while my tests had shown that I could only utilize 2 of them independently, while finally this library seems to be offering all GPIO pins on the board to produce PWM signals.

I'm dazzled with confusion. I'm trying to build a NodeJS quadcopter (so 4 PWM controlled brushless motors) with decent performance and code clarity to be future extension capable.

If you could please help me understand those differing PWM approaches I'll be very thankful...

shtaif, 2016-09-21:
BTW, sorry if it came out critical/offensive, criticism wasn't my intention at all, but a call for help understanding these different mechanisms, their pros and cons.

tralves, 2016-09-21:
I am no expert here. Just adding my 2 cent.

I don't have a RPI 3, but didn't find any documentation stating that it has 4 PWM pins, only 2. You will need as much responsiveness and preciseness as possible to control the quadcopter motors, so software PWM (which this lib provides) probably won't work...

I think that for your case you'll want an external driver such as the PCA9685: https://www.adafruit.com/products/815.

Also, I would advice to check out the johnny-five.io project. Has lots of examples and provides a clearer API.

fivdi, 2016-09-24:
### Hardware PWM

The PWM controller on all variants of the Raspberry Pi has two independent channels. Each channel is available on multiple GPIOs (see section 9.5 on page 140 of [BCM2835 ARM Peripherals](https://www.raspberrypi.org/wp-content/uploads/2012/02/BCM2835-ARM-Peripherals.pdf)):

GPIO12  PWM channel 0
GPIO13  PWM channel 1
GPIO18  PWM channel 0
GPIO19  PWM channel 1
GPIO40  PWM channel 0
GPIO41  PWM channel 1
GPIO45  PWM channel 1
GPIO52  PWM channel 0
GPIO53  PWM channel 1

On a Raspberry Pi 3, GPIO12, GPIO13, GPIO18, and GPIO19 are broken out to the GPIO header. This is why four hardware PWM pins are available on the Raspberry Pi 3. Note that all GPIOs which share a PWM channel will use the same frequency and duty cycle. 

[hardwarePwmWrite](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#hardwarepwmwritefrequency-dutycycle) can be use to control hardware PWM on these GPIOs.

### Hardware Timed PWM

hardwared timed PWM is available on any of GPIO0 through GPIO31. The pulses are timed using the DMA and PWM/PCM hardware peripherals. Hardware timed PWM may not be as precise as hardware PWM but it's still very good. This is NOT the same as software PWM. Software PWM on the Raspberry Pi provides very poor and imprecise results.

[pwmFrequency](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#pwmfrequencyfrequency) and [pwmWrite](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#pwmwritedutycycle) can be used to control hardware timed PWM on any of GPIO0 through GPIO31

@tralves 

> so software PWM (which this lib provides) probably won't work...

This module does NOT provide software PWM. It provides hardware PWM on all pins where it's available and hardware timed PWM on GPIO0 through GPIO31. Hardware PWM and hardware timed PWM are not the same thing as software PWM. Whether it fulfills the requirements @shtaif has is a different question. I'd give it a try.

> Also, I would advice to check out the johnny-five.io project. Has lots of examples and provides a clearer API.

The goal of the pigpio module was to provide an API that's a complete as possible rather than as clear as possible.

fivdi, 2016-09-29:
@shtaif hopefully things are a clearer now. There's nothing actionable here so I'm going to close this one.

closed 2016-09-29 by fivdi

shtaif, 2016-10-03:
I've been researching this PWM functionality mystery for some time (except looking in the Pi's hardware datasheets because I don't understand a thing written there),
but @fivdi just nailed it for me, the best answer I could get.

Actually, to be on the safe side I ordered this external PWM driver:
https://www.adafruit.com/products/2327

However, before I assemble it I'm gonna attempt relying on pigpio's hardware-timed PWM first and see how the motor control turns out.

shtaif, 2016-10-03:
For clarification, the hardware-timed PWM does consume a constant amount of CPU percentage as long as it's active, right?

Pluscrafter, 2018-08-27:
yes and hardware-timed PWM isn't accurate as HardwarePWM.