fivdi/pigpio #10: Library ignores any signal handlers attached before it has initialized

shtaif, 2016-09-21:
#6 

Consider this code sample borrowed from the issue referenced (a little change in the SIGINT handler):

```
var Gpio = require('pigpio').Gpio,
  led = new Gpio(17, {mode: Gpio.OUTPUT}),
  iv;

var iv = setInterval(function () {
  led.digitalWrite(led.digitalRead() ^ 1);
}, 1000);

process.on('SIGINT', function () {
  process.exit();
});
```

Now, if we were to put the SIGINT (or any other signal, as my tests showed) handler _before_ the library module gets `require()`ed:

```
process.on('SIGINT', function () {
  process.exit();
});

var Gpio = require('pigpio').Gpio,
  led = new Gpio(17, {mode: Gpio.OUTPUT}),
  iv;

var iv = setInterval(function () {
  led.digitalWrite(led.digitalRead() ^ 1);
}, 1000);
```

Then we still get the `sigHandler: Unhandled signal 2, terminating`

It took me quite some time to debug this out throughout my code, because I've got a custom signal handler inside a seperate file, which is required in different files which attach more functions to it before it calls `process.exit`. And, the first time my signal handler is being `require()`ed (and doing `process.on('SIGINT', ...)`), is before the first time pigpio gets `require()`ed, just like the second code example I put, so perhaps the pigpio's initialization is doing something not properly.

This issue is more than just avoiding the `sigHandler: Unhandled signal 2, terminating` - the scenario I described, when is performed using, for instance, [nodemon](https://github.com/remy/nodemon), caused some even more serious errors to be output.

Hope I made it clear enough,
Shtaif.

fivdi, 2016-09-24:
On Linux, only one signal handler can be registered for calling. If multiple signal handlers are registered, only the last handler registered will be called. In both of the examples above, two signals handlers are registered, but only the last one registered is called.

The following line of code in both examples registers a handler:

```js
led = new Gpio(17, {mode: Gpio.OUTPUT}),
```

This line of code results in the underlying [pigpio C library](https://github.com/joan2937/pigpio) being initialized. During the initialization, the pigpio C library library registers a `SIGINT` signal handler.

In the first example above, the JavaScript handler is the last handler registered so it will be called.

In the second example, the pigpio C library is the last to register a handler so that gets called.

I don't think there is anything the pigpio JavaScript module can do to change things here.

fivdi, 2016-09-25:
pigpio v0.4.0 has just been released and should make it possible to improve things here. `initialize` and `terminate` methods have been added to the pigpio API. See https://github.com/fivdi/pigpio/blob/master/doc/configuration.md for further details.

fivdi, 2016-09-29:
@shtaif does the modification to pigpio help?

shtaif, 2016-10-03:
@fivdi yes, I neatly implemented the two methods you added and all seems green.
Thanks, this made the library a lot more **useful and flexible**!

closed 2016-10-03 by shtaif