fivdi/pigpio #11: fork in node causes error on the second child

hamiltonmilligan, 2016-10-03:
running a script where
const child_process = require('child_process');
var kicker_process = child_process.fork(`${__dirname}/botjs/ki.js`); //
in ki.js we have
var pigpio = require('pigpio'),  pig= pigpio.Gpio
ki_process = new Ki(pig,18);
ki_process.init();

all works good. But if you have a second child_process.fork with a 
 var pigpio = require('pigpio'),  pig= pigpio.Gpio
in it.  it fails in initallising it. pigpio line 11.
It should work. 
I have many sensors that have their own js file. all plan to run in a child_process fork. All planning to use pigpio. I can only run one child process that uses pigpio in it at the moment.

fivdi, 2016-10-03:
This is a restriction of the pigpio C library which is wrapped by the pigpio JavaScript package. It's not possible to use the pigpio C library in two processes at the same time.

fivdi, 2016-10-15:
Closing as there's nothing that can be done in the pigpio Node.js module to resolve this issue.

closed 2016-10-15 by fivdi