fivdi/pigpio #13: npm install error

haldunatar, 2016-10-28:
As I run npm i pigpio I get the following error:

 ...
319 verbose about to build /home/pi/Desktop/sensorjs/node_modules/pigpio/node_modules/nan
320 info build /home/pi/Desktop/sensorjs/node_modules/pigpio/node_modules/nan
321 info linkStuff nan@2.4.0
322 silly linkStuff nan@2.4.0 has /home/pi/Desktop/sensorjs/node_modules/pigpio/node_modules as its parent node_modules
323 verbose linkBins nan@2.4.0
324 verbose linkMans nan@2.4.0
325 verbose rebuildBundles nan@2.4.0
326 info install nan@2.4.0
327 info postinstall nan@2.4.0
328 verbose unlock done using /home/pi/.npm/_locks/nan-541513cab34103a0.lock for /home/pi/Desktop/sensorjs/node_modules/pigpio/node_modules/nan
329 verbose about to build /home/pi/Desktop/sensorjs/node_modules/pigpio
330 info build /home/pi/Desktop/sensorjs/node_modules/pigpio
331 info linkStuff pigpio@0.4.0
332 silly linkStuff pigpio@0.4.0 has /home/pi/Desktop/sensorjs/node_modules as its parent node_modules
333 verbose linkBins pigpio@0.4.0
334 verbose linkMans pigpio@0.4.0
335 verbose rebuildBundles pigpio@0.4.0
336 verbose rebuildBundles [ 'bindings', 'nan' ]
337 info install pigpio@0.4.0
338 verbose unsafe-perm in lifecycle true
339 info pigpio@0.4.0 Failed to exec install script
340 verbose unlock done using /home/pi/.npm/_locks/pigpio-341b68c5be5148f0.lock for /home/pi/Desktop/sensorjs/node_modules/pigpio
341 verbose stack Error: pigpio@0.4.0 install: `node-gyp rebuild`
341 verbose stack Exit status 1
341 verbose stack     at EventEmitter.<anonymous> (/home/pi/.nvm/versions/node/v4.0.0/lib/node_modules/npm/lib/utils/lifecycle.js:214:16)
341 verbose stack     at emitTwo (events.js:87:13)
341 verbose stack     at EventEmitter.emit (events.js:172:7)
341 verbose stack     at ChildProcess.<anonymous> (/home/pi/.nvm/versions/node/v4.0.0/lib/node_modules/npm/lib/utils/spawn.js:24:14)
341 verbose stack     at emitTwo (events.js:87:13)
341 verbose stack     at ChildProcess.emit (events.js:172:7)
341 verbose stack     at maybeClose (internal/child_process.js:817:16)
341 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
342 verbose pkgid pigpio@0.4.0
343 verbose cwd /home/pi/Desktop/sensorjs
344 error Linux 4.1.18-v7+
345 error argv "/home/pi/.nvm/versions/node/v4.0.0/bin/node" "/home/pi/.nvm/versions/node/v4.0.0/bin/npm" "i" "pigpio"
346 error node v4.0.0
347 error npm  v2.14.2
348 error code ELIFECYCLE
349 error pigpio@0.4.0 install: `node-gyp rebuild`
349 error Exit status 1
350 error Failed at the pigpio@0.4.0 install script 'node-gyp rebuild'.
350 error This is most likely a problem with the pigpio package,
350 error not with npm itself.
350 error Tell the author that this fails on your system:
350 error     node-gyp rebuild
350 error You can get their info via:
350 error     npm owner ls pigpio
350 error There is likely additional logging output above.
351 verbose exit [ 1, true ]
352 verbose unbuild node_modules/pigpio
353 info preuninstall pigpio@0.4.0
354 info uninstall pigpio@0.4.0
355 verbose unbuild rmStuff pigpio@0.4.0 from /home/pi/Desktop/sensorjs/node_modules
356 info postuninstall pigpio@0.4.0
357 silly gentlyRm /home/pi/Desktop/sensorjs/node_modules/pigpio is being purged from base /home/pi/Desktop/sensorjs
358 verbose gentlyRm don't care about contents; nuking /home/pi/Desktop/sensorjs/node_modules/pigpio
359 silly vacuum-fs purging /home/pi/Desktop/sensorjs/node_modules/pigpio
360 silly vacuum-fs removing /home/pi/Desktop/sensorjs/node_modules
361 silly vacuum-fs finished vacuuming up to /home/pi/Desktop/sensorjs

any idea?

fivdi, 2016-10-28:
That looks like the contents of npm-debug.log and doesn't contain much information of interest. Please post the complete output of npm i pigpio, i.e., all information printed to the screen during the install process.

haldunatar, 2016-10-29:
pi@raspberrypi:~/Desktop/sensorjs $ npm i pigpio
npm WARN package.json sensorjs@1.0.0 No description
npm WARN package.json sensorjs@1.0.0 No repository field.
npm WARN package.json sensorjs@1.0.0 No README data
|
> pigpio@0.4.0 install /home/pi/Desktop/sensorjs/node_modules/pigpio
> node-gyp rebuild

make: Entering directory '/home/pi/Desktop/sensorjs/node_modules/pigpio/build'
  CXX(target) Release/obj.target/pigpio/src/pigpio.o
../src/pigpio.cc:2:20: fatal error: pigpio.h: No such file or directory
 #include <pigpio.h>
                    ^
compilation terminated.
pigpio.target.mk:90: recipe for target 'Release/obj.target/pigpio/src/pigpio.o' failed
make: *** [Release/obj.target/pigpio/src/pigpio.o] Error 1
make: Leaving directory '/home/pi/Desktop/sensorjs/node_modules/pigpio/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/.nvm/versions/node/v4.0.0/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.1.18-v7+
gyp ERR! command "/home/pi/.nvm/versions/node/v4.0.0/bin/node" "/home/pi/.nvm/versions/node/v4.0.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/Desktop/sensorjs/node_modules/pigpio
gyp ERR! node -v v4.0.0
gyp ERR! node-gyp -v v3.0.1
gyp ERR! not ok 
npm ERR! Linux 4.1.18-v7+
npm ERR! argv "/home/pi/.nvm/versions/node/v4.0.0/bin/node" "/home/pi/.nvm/versions/node/v4.0.0/bin/npm" "i" "pigpio"
npm ERR! node v4.0.0
npm ERR! npm  v2.14.2
npm ERR! code ELIFECYCLE

npm ERR! pigpio@0.4.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the pigpio@0.4.0 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the pigpio package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls pigpio
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/Desktop/sensorjs/npm-debug.log

fivdi, 2016-10-29:
It looks like the pigpio C library hasn't been installed. It's a prepequsite for the pigpio Node.js module. Newer versions of Raspbian come with the pigpio C library preinstalled. On older versions of Raspbian the pigpio C library needs to be manually installed. See installation step 1 at https://github.com/fivdi/pigpio#installation

haldunatar, 2016-10-29:
Actually, I had installed the C library. Double checked and I saw that I didn't install it in the root and that was the problem. I have installed it in correct path and now it works! Thank you!

closed 2016-10-29 by haldunatar