fivdi/pigpio #14: is there any thing by which we can read sensor value

DaniyalGeek, 2016-11-13:


fivdi, 2016-11-13:
What type of sensor would you like to read a value from? If possible specify the exact sensor type.

DaniyalGeek, 2016-11-13:
LM35 temperature sensor

fivdi, 2016-11-13:
The LM35 temperature sensor has an analog output but the Raspberry Pi doesn't have analog inputs. This means that additional hardware to perform analog to digital conversion is needed. In addition, the LM35 operates at 4V to 30V and this also needs to be taken into account.

For analog to digital conversion an MCP3008 could be used. The [mcp-spi-adc](https://github.com/fivdi/mcp-spi-adc) module could then be used to interface with the MCP3008 from Node.js.

For information about operating the MCP3008 and LM35 at 5V see [here](http://raspberrypi.stackexchange.com/questions/40065/raspberry-pi-lm35-and-mcp3008).

fivdi, 2016-11-13:
If it's not essential to use an LM35 temperature sensor a TMP36 temperature sensor could be used instead. This would be slightly easier and is described at [mcp-spi-adc](https://github.com/fivdi/mcp-spi-adc).

DaniyalGeek, 2016-11-13:
Could you tell me about coding which function i can use i am new with raspberry pi

fivdi, 2016-11-13:
[here's](https://github.com/fivdi/mcp-spi-adc#usage) a complete example for the TMP36. The example would have to be modified for the LM35 but shouldn't be too difficult to figure out.

fivdi, 2016-11-13:
I'm going to close this one as there's no actionable for pigpio.

closed 2016-11-13 by fivdi