fivdi/pigpio #16: node-gyp rebuild fails

Raqbit, 2017-03-01:
Error: http://pastebin.com/ztq29Ekb

**Node version**: v7.7.0 (Installed manually)

Getting this error when  trying to install this package. It might be because of my non-ordinary node install method.

fivdi, 2017-03-01:
Very nice, the newest of software on the newest of hardware :)

This is a Node.js issue. V7.0.0 is broken. See https://github.com/nodejs/nan/issues/644 for further information. 

It looks like you'll have to downgraded to an older version of Node.is or wait for V7.0.1.

fivdi, 2017-03-01:
The corresponding Node.js issue is https://github.com/nodejs/node/issues/11628

fivdi, 2017-03-02:
Node.js v7.7.1 has been released so this should no longer be an issue

closed 2017-03-02 by fivdi

Raqbit, 2017-03-02:
Oh right, I should've noticed that it wasn't the latest LTS >.< Thanks for the help anyways!