fivdi/pigpio #17: [FR] swith to node-pre-gyp

farfromrefug, 2017-03-07:
That would be awesome if you could switch from node-gyp to [node-pre-gyp](https://github.com/mapbox/node-pre-gyp).
that way we could have pre build release and make ```npm install``` so much faster!
If you need an example [node-usb](https://github.com/tessel/node-usb) uses it

Thanks

fivdi, 2017-03-07:
pigpio is for all variants of the Raspberry Pi all of which have ARM processors. To the best of my knowledge there are no node modules that use node-pre-gyp with binaries for ARM architectures. I've only ever seen it being used for binaries for ia32 and x64 Intel architectures. The binaries available for some popular modules are, for example, those for [leveldown](https://github.com/Level/leveldown/releases/tag/v1.6.0), [node-serialport](https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/tag/4.0.7) and [usb](https://github.com/tessel/node-usb/releases/tag/1.2.0). As can be seen, binaries are available for the ia32 and x64 architectures but not for any of the ARM archictures. It can also be seen that there are thirty different binaries for a single version of serialport. Note that there are quite a number of ARM architectures, there's ARMv6, ARMv7-A, ARMv8 (32 & 64 bit), ...

To be honest, given the large effort that would be required to implement this feature, it's not something that I'm prepared to work on. Also, given that I haven't seen node-pre-gyp being used for binaries for any of the ARM architectures, I'm even less inclined to work on the topic.

In additon, it only takes a little more than 12 seconds to install pigpio on a Raspberry Pi 3. This may not be blazing fast, but it's not bad.
```
pi@raspberrypi:~/node-v7.7.1 $ time npm install pigpio

> pigpio@0.4.0 install /home/pi/node-v7.7.1/node_modules/pigpio
> node-gyp rebuild

make: Entering directory '/home/pi/node-v7.7.1/node_modules/pigpio/build'
  CXX(target) Release/obj.target/pigpio/src/pigpio.o
  SOLINK_MODULE(target) Release/obj.target/pigpio.node
  COPY Release/pigpio.node
make: Leaving directory '/home/pi/node-v7.7.1/node_modules/pigpio/build'
/home/pi/node-v7.7.1
└─┬ pigpio@0.4.0 
  ├── bindings@1.2.1 
  └── nan@2.4.0 

npm WARN enoent ENOENT: no such file or directory, open '/home/pi/node-v7.7.1/package.json'
npm WARN node-v7.7.1 No description
npm WARN node-v7.7.1 No repository field.
npm WARN node-v7.7.1 No README data
npm WARN node-v7.7.1 No license field.

real	0m12.502s
user	0m11.990s
sys	0m0.620s
pi@raspberrypi:~/node-v7.7.1 $ 
```

Sorry!

closed 2017-03-07 by fivdi

farfromrefug, 2017-03-08:
@fivdi that s sad. And the effort is very small. Also you might not see binaries for usb, serialport ... but it allow anyone to actually host his own binaries on his website! Which is huge for production.

I will try to make a pr

fivdi, 2017-03-11:
> @fivdi that s sad.

Indeed, but I did say sorry before closing the issue :)

> And the effort is very small.

Maybe this is a point where we should agree to disagree. Here are some areas where I can imagine quite a bit of effort being needed:

### Automatically generating all the required binaries
Different versions of the Raspberry Pi are based on different ARM architectures. ARMv6 for the Pi Zero and 1, ARMv7 for the Pi2 and ARMv8 for the Pi 3. For ARMv8 there's 32bit and 64bit variants. This means that binaries are needed for four different variants of the ARM architecture. On top of this, each version of [Node.js has a different ABI](https://nodejs.org/en/download/releases/) which means that different binaries are also needed for each version of Node.js that's supported. Currently Node.js v0.10, v0.12, v4 and v6 are supported by pigpio. This means that 4 x 4 = 16 different binaries are needed for every single release of pigpio. Note that this is only for the Raspberry Pi. This feature was also [requested for i2c-bus](https://github.com/fivdi/i2c-bus/issues/33). What if i2c-bus is installed on an Intel Galileo Gen 2 with the ia32 architecture that's different to the ia32 architecture on PCs? (See https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/747 for further details)

It wouldn't make sense to generate all these binaries manually, they would have to be generated automatically. Where are the resources for generating all these binaries automatically?

### Testing the generated binaries

Given that a large number of binaries are needed it doesn't make sense to test them manually, they would have to be tested automatically. Where are the resources for automatically testing all the binaries?  What tests need to be automatically performed? Have these tests already been implemented? (I think not.)

### Additional Question
 * Why do all the packages currently using node-pre-gyp only have pre-built binaries for the Intel ia32 and x64 architectures and not for any of the ARM architectures?

farfromrefug, 2017-03-13:
@fivdi you did not understand my point. I totally agree with you on what you said.
The idea is not to have your repo maintain the binaries/releases. Node-pre-gyp has a great feature called "host override" which allows anyone to provide it's own custom binary repo while running "npm install".
So you don't have to host any release for any platform! By using ```node-pre-gyp``` you just allow anyone to:
- use a community driven and maintained module
- provide its own binaries for the platforms it uses

In my test case it's very practical to use that method. I can have a dev version of each platform where i build the binaries, and then on all our "products" i just have to override the host and thus i am sure that the build will work and that the product will have the version i want.
Also something i can't do right now, i could even remove all "compilation" stuff from our products. It would make the product distribution even smaller.

Once again i totally agree with all your points. I will try to make a PR and then you can merge it if you think it's a good addition.