fivdi/pigpio #18: Change name to avoid confusion?

cymplecy, 2017-03-13:
Just a susggestion but maybe better to call this pigpioj or something like that.  Calling it pigpio can cause confusion - at least it did for me when someone suggested using it on the Pi forum :)

fivdi, 2017-03-13:
Thank you for the suggestion.
With which other projects did you see the confusion and which project were you actually searching for?

cymplecy, 2017-03-13:
The actual pigpio library itself.
I was reading a Pi forum thread and someone directed a question to your Node.js project called pigpio - and I thought - hang on - joan's pigpio isn't javascript! :)  When I got here I realised you've done a wrapper for the underlying pigpio library.  So I thought - to save someone else making same mistook as me - I'd make the suggestion  :)

cymplecy, 2017-03-13:
Great work BTW :)

fivdi, 2017-03-13:
Hmm, I don't want to change the name of the project. On the other hand, there is potential for confusion. To reduce the potential for confusion, the [description of the project at the top of the readme](https://github.com/fivdi/pigpio/blob/master/README.md#pigpio) could explicitly state that the pigpio JavaScript module is a wrapper for the pigpio C library.

cymplecy, 2017-03-13:
Last thoughts :)  If you had written a piece of software from scratch called foobar and if it was often used and referred to by many people and well known as foobar and then someone made a modification for it but instead of calling it wrapper for foobar - they just called it foobar - how would you feel about that? :) Entirely up to you - I'll say no more :)

fivdi, 2017-03-13:
> Last thoughts :) If you had written a piece of software from scratch called foobar and if it was often used and referred to by many people and well known as foobar and then someone made a [...]

Oh I'd regard that as a complement and feel satisfaction with myself. It would give me a sense of accomplishment :)

> Entirely up to you

Correct :)

> I'll say no more :)

That's nice to hear :)

fivdi, 2017-03-13:
Project description improved with 281773c0031a06d2a1fad97dd32ca05ae9a9173c

@cymplecy Thank you for your contribution. Hopefully there will be less confusion in the future :)

closed 2017-03-13 by fivdi