fivdi/pigpio #19: Script crash with error after rpi3 reboot.

Yoda49, 2017-04-22:
```
/home/antio/smarthome/node_modules/bindings/bindings.js:83
        throw e
              ^
Error: /home/antio/smarthome/node_modules/pigpio/build/Release/pigpio.node: undefined symbol: node_module_register
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at bindings (/home/antio/smarthome/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/home/antio/smarthome/node_modules/pigpio/pigpio.js:5:31)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
error: Forever detected script exited with code: 8
error: Script restart attempt #133
```

fivdi, 2017-04-22:
Was the system upgraded to a new version of Node.js or downgraded to an old version of Node.js?

If so, pigpio may have to be recompiled as it's a native module and native modules need to be recompiled when the [Node.js ABI version number changes (NODE_MODULE_VERSION)](https://nodejs.org/en/download/releases/). The easiest was to recompile pigpio is to install it again.

fivdi, 2017-04-24:
@Yoda49 was it possible to make any progress here?

Yoda49, 2017-04-26:
There was no time to try. Today i will try.

fivdi, 2017-04-26:
It looks like the system isn't configured correctly to run applications at boot time using Node.js v7.7.4.

The top three lines of the stack trace reference line 356, 312 and 364 in module.js. Here are links to those lines in module.js in Node.js v7.7.4:

Error: /home/antio/smarthome/node_modules/pigpio/build/Release/pigpio.node: undefined symbol: node_module_register
    at Module.load ([module.js:356:32](https://github.com/nodejs/node/blob/v7.7.4/lib/module.js#L356))
    at Function.Module._load ([module.js:312:12](https://github.com/nodejs/node/blob/v7.7.4/lib/module.js#L312))
    at Module.require ([module.js:364:17](https://github.com/nodejs/node/blob/v7.7.4/lib/module.js#L364))

Lines 356 and 312 are comments and line 364 is a break. Lines that are comments can't appear in a stack trace as they don't contain code. This implies that Node.js v7.7.4 is not used to execute the application when it's started at boot time.

There must be another version of Node.js installed and that's being used for the application when it's started at boot time. Raspbian Jessie comes with Node.js v0.10.29 pre-installed. If I had to guess, I'd say that Node.js v0.10.29 is being used when the application is started at boot time.

Here are links to the same lines in Node.js v0.10.29 and they also correspond to what the exception is saying:

Error: /home/antio/smarthome/node_modules/pigpio/build/Release/pigpio.node: undefined symbol: node_module_register
    at Module.load ([module.js:356:32](https://github.com/nodejs/node/blob/v0.10.29/lib/module.js#L356))
    at Function.Module._load ([module.js:312:12](https://github.com/nodejs/node/blob/v0.10.29/lib/module.js#L312))
    at Module.require ([module.js:364:17](https://github.com/nodejs/node/blob/v0.10.29/lib/module.js#L364))

The issue isn't related to pigpio, it's related to how Node.js is configured on the system.

Yoda49, 2017-04-27:
"nvm ls" listing:

```
root@RPI3:/home/antio# nvm ls
->       v7.7.4
         system
default -> v7.7.4
node -> stable (-> v7.7.4) (default)
stable -> 7.7 (-> v7.7.4) (default)
iojs -> N/A (default)
lts/* -> lts/boron (-> N/A)
lts/argon -> v4.8.2 (-> N/A)
lts/boron -> v6.10.2 (-> N/A)
root@RPI3:/home/antio#
```

fivdi, 2017-04-27:
nvm may list that for user antio after they logon but that's not necessarily the version used at boot time.

If the following lines are added to the application, what are the contents of `app.log`?

```
var fs=require("fs");
fs.writeFileSync("app.log", process.versions.node);
```

Yoda49, 2017-04-27:
Yes, You are right. I am sorry.  In my system there are two version of nodejs.

```
root@RPI3:~/smarthome# whereis forever
forever: /usr/local/bin/forever /root/.nvm/versions/node/v7.7.4/bin/forever
```
When system reboot, forever run automatically from **/usr/local/bin/forever** in rc.local. But after reboot when I manually start forever, it run from **/root/.nvm/versions/node/v7.7.4/bin/forever**. I don't understand why so =/

fivdi, 2017-04-27:
Perhaps it's best to uninstall the old stuff if it's not needed.

Yoda49, 2017-04-27:
The problem is solved.

fivdi, 2017-04-27:
That's good news :)

closed 2017-04-27 by fivdi