fivdi/pigpio #20: node-gyp build fails

dkebler, 2017-05-17:
Built pigpio(d) version 64 successfully and it passes all the tests provided.

But installing js pigpio throws all kind of nan errors.
using  node 7.10, npm 4.2, ubuntu mate 16.04 on a RPI3

```
In file included from ../../nan/nan.h:194:0,
                 from ../src/pigpio.cc:2:
../../nan/nan_maybe_43_inl.h: In function ‘Nan::MaybeLocal<v8::Object> Nan::CloneElementAt(v8::Local<v8::Array>, uint32_t)’:
../../nan/nan_maybe_43_inl.h:221:58: warning: ‘v8::MaybeLocal<v8::Object> v8::Array::CloneElementAt(v8::Local<v8::Context>, uint32_t)’ is deprecated: Cloning is not supported. [-Wdeprecated-declarations]
   return array->CloneElementAt(GetCurrentContext(), index);
                                                          ^
In file included from /home/sysadmin/.node-gyp/7.10.0/include/node/v8.h:26:0,
                 from /home/sysadmin/.node-gyp/7.10.0/include/node/node.h:42,
                 from ../../nan/nan.h:45,
                 from ../src/pigpio.cc:2:
/home/sysadmin/.node-gyp/7.10.0/include/node/v8.h:3347:36: note: declared here
                 MaybeLocal<Object> CloneElementAt(Local<Context> context,
                                    ^
/home/sysadmin/.node-gyp/7.10.0/include/node/v8config.h:329:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^
In file included from ../../nan/nan_new.h:189:0,
                 from ../../nan/nan.h:200,
                 from ../src/pigpio.cc:2:
../../nan/nan_implementation_12_inl.h: In static member function ‘static Nan::imp::FactoryBase<v8::BooleanObject>::return_t Nan::imp::Factory<v8::BooleanObject>::New(bool)’:
../../nan/nan_implementation_12_inl.h:40:38: warning: ‘static v8::Local<v8::Value> v8::BooleanObject::New(bool)’ is deprecated: Pass an isolate [-Wdeprecated-declarations]
   return v8::BooleanObject::New(value).As<v8::BooleanObject>();
                                      ^
In file included from /home/sysadmin/.node-gyp/7.10.0/include/node/v8.h:26:0,
                 from /home/sysadmin/.node-gyp/7.10.0/include/node/node.h:42,
                 from ../../nan/nan.h:45,
                 from ../src/pigpio.cc:2:
/home/sysadmin/.node-gyp/7.10.0/include/node/v8.h:4490:56: note: declared here
   V8_DEPRECATED("Pass an isolate", static Local<Value> New(bool value));
                                                        ^
/home/sysadmin/.node-gyp/7.10.0/include/node/v8config.h:329:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^
In file included from ../src/pigpio.cc:2:0:
../../nan/nan.h: At global scope:
../../nan/nan.h:602:20: error: variable or field ‘AddGCEpilogueCallback’ declared void
       v8::Isolate::GCEpilogueCallback callback
                    ^
../../nan/nan.h:602:7: error: ‘GCEpilogueCallback’ is not a member of ‘v8::Isolate’
       v8::Isolate::GCEpilogueCallback callback
       ^
../../nan/nan.h:603:18: error: expected primary-expression before ‘gc_type_filter’
     , v8::GCType gc_type_filter = v8::kGCTypeAll) {
                  ^
../../nan/nan.h:608:20: error: variable or field ‘RemoveGCEpilogueCallback’ declared void
       v8::Isolate::GCEpilogueCallback callback) {
                    ^
../../nan/nan.h:608:7: error: ‘GCEpilogueCallback’ is not a member of ‘v8::Isolate’
       v8::Isolate::GCEpilogueCallback callback) {
       ^
../../nan/nan.h:613:20: error: variable or field ‘AddGCPrologueCallback’ declared void
       v8::Isolate::GCPrologueCallback callback
                    ^
../../nan/nan.h:613:7: error: ‘GCPrologueCallback’ is not a member of ‘v8::Isolate’
       v8::Isolate::GCPrologueCallback callback
       ^
../../nan/nan.h:614:18: error: expected primary-expression before ‘gc_type_filter’
     , v8::GCType gc_type_filter = v8::kGCTypeAll) {
                  ^
../../nan/nan.h:619:20: error: variable or field ‘RemoveGCPrologueCallback’ declared void
       v8::Isolate::GCPrologueCallback callback) {
                    ^
../../nan/nan.h:619:7: error: ‘GCPrologueCallback’ is not a member of ‘v8::Isolate’
       v8::Isolate::GCPrologueCallback callback) {
       ^
../../nan/nan.h: In function ‘bool Nan::SetAccessor(v8::Local<v8::Object>, v8::Local<v8::String>, Nan::GetterCallback, Nan::SetterCallback, v8::Local<v8::Value>, v8::AccessControl, v8::PropertyAttribute)’:
../../nan/nan.h:1961:16: warning: ‘bool v8::Object::SetAccessor(v8::Local<v8::Name>, v8::AccessorNameGetterCallback, v8::AccessorNameSetterCallback, v8::Local<v8::Value>, v8::AccessControl, v8::PropertyAttribute)’ is deprecated: Use maybe version [-Wdeprecated-declarations]
     , attribute);
                ^
In file included from /home/sysadmin/.node-gyp/7.10.0/include/node/v8.h:26:0,
                 from /home/sysadmin/.node-gyp/7.10.0/include/node/node.h:42,
                 from ../../nan/nan.h:45,
                 from ../src/pigpio.cc:2:
/home/sysadmin/.node-gyp/7.10.0/include/node/v8.h:3027:22: note: declared here
                 bool SetAccessor(Local<Name> name,
                      ^
/home/sysadmin/.node-gyp/7.10.0/include/node/v8config.h:329:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^
pigpio.target.mk:94: recipe for target 'Release/obj.target/pigpio/src/pigpio.o' failed
make: *** [Release/obj.target/pigpio/src/pigpio.o] Error 1
make: Leaving directory '/home/sysadmin/uci/uci-examples/node_modules/uci-interrupt/node_modules/pigpio/build'
```

dkebler, 2017-05-17:
I did have the daemon running so per your instructions I stopped it, but building again I get the same errors

fivdi, 2017-05-17:
Can you post the exact command used to perform the install and the complete output of that command please?

fivdi, 2017-05-17:
I would imagine that an attempt is being made to install an old version of pigpio. Please try with the latest version of pigpio which is v0.5.1.

dkebler, 2017-05-18:
I assume you mean pigpio since the above is complete output from npm when running `npm install pigpio` 

I did...
```
wget abyz.co.uk/rpi/pigpio/pigpio.tar
tar xf pigpio.tar
cd PIGPIO
make
sudo make install
```
from pigpio site.  It does throw some warnings.  See below

I found there was some pigpio loaded from the orginial image says 1.30-1-xenial??  so I purged that 

What exact files  are you using in what linux directory  Is is /usr/local/???.   Only the daemon has a cli that reports a version and it says version 61 so kinda hard to determine the version after install.

this is what sudo make install says
```
install -m 0755 -d                /opt/pigpio/cgi
install -m 0755 -d                /usr/local/include
install -m 0644 pigpio.h          /usr/local/include
install -m 0644 pigpiod_if.h      /usr/local/include
install -m 0644 pigpiod_if2.h     /usr/local/include
install -m 0755 -d                /usr/local/lib
install -m 0755 libpigpio.so      /usr/local/lib
install -m 0755 libpigpiod_if.so  /usr/local/lib
install -m 0755 libpigpiod_if2.so /usr/local/lib
install -m 0755 -d                /usr/local/bin
install -m 0755 pig2vcd           /usr/local/bin
install -m 0755 pigpiod           /usr/local/bin
install -m 0755 pigs              /usr/local/bin
```

make output
```
sysadmin@sbc:/opt/PIGPIO$ make
gcc -O3 -Wall -pthread -fpic -c -o pigpio.o pigpio.c
pigpio.c: In function ‘pthTimerTick’:
pigpio.c:6764:29: warning: format not a string literal and no format arguments [-Wformat-security]
             fprintf(stderr, buf);
                             ^
pigpio.c: In function ‘pthFifoThread’:
pigpio.c:6855:19: warning: format not a string literal and no format arguments [-Wformat-security]
                   fprintf(outFifo, cmdUsage);
                   ^
pigpio.c: In function ‘i2cOpen’:
pigpio.c:3979:7: warning: ignoring return value of ‘system’, declared with attribute warn_unused_result [-Wunused-result]
       system("/sbin/modprobe i2c_dev");
       ^
pigpio.c:3980:7: warning: ignoring return value of ‘system’, declared with attribute warn_unused_result [-Wunused-result]
       system("/sbin/modprobe i2c_bcm2835");
       ^
pigpio.c: In function ‘i2cSwitchCombined’:
pigpio.c:4042:20: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
       if (setting) write(fd, "1\n", 2); else write(fd, "0\n", 2);
                    ^
pigpio.c:4042:46: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
       if (setting) write(fd, "1\n", 2); else write(fd, "0\n", 2);
                                              ^
pigpio.c: In function ‘pthSocketThreadHandler’:
pigpio.c:6975:7: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
       write(sock, p, 16);
       ^
pigpio.c:7000:16: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
                write(sock, buf, p[3]);
                ^
pigpio.c: In function ‘initGrabLockFile’:
pigpio.c:7152:10: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
          write(fd, pidStr, strlen(pidStr));
          ^
pigpio.c: In function ‘pthISRThread’:
pigpio.c:11345:4: warning: ignoring return value of ‘read’, declared with attribute warn_unused_result [-Wunused-result]
    read(fd, buf, sizeof buf);
    ^
pigpio.c:11358:10: warning: ignoring return value of ‘read’, declared with attribute warn_unused_result [-Wunused-result]
          read(fd, buf, sizeof buf);
          ^
gcc -O3 -Wall -pthread -fpic -c -o command.o command.c
gcc -shared -o libpigpio.so pigpio.o command.o
strip --strip-unneeded libpigpio.so
size     libpigpio.so
   text	   data	    bss	    dec	    hex	filename
 181116	   5360	 598336	 784812	  bf9ac	libpigpio.so
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if.o pigpiod_if.c
gcc -shared -o libpigpiod_if.so pigpiod_if.o command.o
strip --strip-unneeded libpigpiod_if.so
size     libpigpiod_if.so
   text	   data	    bss	    dec	    hex	filename
  51786	   4316	  49244	 105346	  19b82	libpigpiod_if.so
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if2.o pigpiod_if2.c
gcc -shared -o libpigpiod_if2.so pigpiod_if2.o command.o
strip --strip-unneeded libpigpiod_if2.so
size     libpigpiod_if2.so
   text	   data	    bss	    dec	    hex	filename
  59821	   4316	   1984	  66121	  10249	libpigpiod_if2.so
gcc -O3 -Wall -pthread   -c -o x_pigpio.o x_pigpio.c
gcc -o x_pigpio x_pigpio.o -L. -lpigpio -pthread -lrt
gcc -O3 -Wall -pthread   -c -o x_pigpiod_if.o x_pigpiod_if.c
gcc -o x_pigpiod_if x_pigpiod_if.o -L. -lpigpiod_if -pthread -lrt
gcc -O3 -Wall -pthread   -c -o x_pigpiod_if2.o x_pigpiod_if2.c
gcc -o x_pigpiod_if2 x_pigpiod_if2.o -L. -lpigpiod_if2 -pthread -lrt
gcc -O3 -Wall -pthread   -c -o pig2vcd.o pig2vcd.c
gcc -o pig2vcd pig2vcd.o
strip pig2vcd
gcc -O3 -Wall -pthread   -c -o pigpiod.o pigpiod.c
gcc -o pigpiod pigpiod.o -L. -lpigpio -pthread -lrt
strip pigpiod
gcc -O3 -Wall -pthread   -c -o pigs.o pigs.c
gcc -o pigs pigs.o command.o
strip pigs
```

dkebler, 2017-05-18:
I the the pigpio c library has a version function, but it doesn't look like you provided a binding for that so I can see what version pigpio node thinks it's using?

fivdi, 2017-05-18:
I don't think there's an issue with the pigpio C library. It's the pigpio JS package that's causing the issue. Is that really the complete output of `npm install pigpio`? As in every single character of output?

Also, does the directory where `npm install pigpio` is being called from contain a package.json file? If so, what are the contents of that file?

dkebler, 2017-05-18:
You were right it was the version.   

-----

I am using pigipo as a dependency in a package to the code I am running.   I have a headless setup so I actually code on my main box and it syncs the packages I am working on to node_modules on the main code on the rpi.  So I obviously can't do an install on that machine but rather later with npm install without save on the RPI.  That means I have to hand code the package into package.json.  Not knowing the version I just put in ^0.0.1  assuming npm would pick up the latest....it didn't.  Just my misunderstanding of how npm uses semiversions.  I put in "0.5.1" and all is well....:-).  I better read up on npm semiversion useage again!

BTW I have done this with epoll, and ic2-bus successfully so was kinda scratching my head.

closed 2017-05-18 by dkebler

fivdi, 2017-05-18:
Good to hear it's working now :)