fivdi/pigpio #21: How to use terminate() function

arthurmoises, 2017-05-30:
Hi! Awesome code that you wrote! Thanks a lot for this work!

I'm trying to understand how to use the terminate() function. 

Basically, my "hello world" code is the following one: 

`var led = new Gpio(5, {mode: Gpio.OUTPUT});

setInterval(function(err) {
  if (err) { throw err; }
  led.pwmWrite(127);
}, 200);

process.on('SIGINT', function() {
  led.terminate();
});
`

What I intend to do is to free the variables when I hit "Ctrl+C" on a linux terminal. When I do that, the led is still HIGH, and I have to run the program, hit Ctrl+C for the second time to turn the led off.

All the times that I hit Ctrl+C, I got this error: 

`  led.terminate();
      ^

TypeError: led.terminate is not a function
    at process.<anonymous> (/home/pi/[repo name]/index.js:22:7)
    at emitNone (events.js:86:13)
    at process.emit (events.js:185:7)
    at Signal.wrap.onsignal (internal/process.js:199:44)
`
I'm doing it the right way? It's possible to free the variable (and return the led to the initial state - after the program ends - somehow?) Like the way the 'onoff' unexport() function does?

Thanks a lot in advance!

arthurmoises, 2017-05-30:
Hi! Issue #10 answered my question, sorry for not reading all those answers before posting. You can close this one.

fivdi, 2017-05-30:
Glad you got it working.

Using terminate is tricky so it would be better if there was an example in the documentation showing how to use it. I'll reopen the issue and add an example.

fivdi, 2017-07-12:
The [documentation](https://github.com/fivdi/pigpio/blob/master/doc/configuration.md#initialize) has been improved and now has an example showing how to use the `initialize` and `terminate` functions.

closed 2017-07-12 by fivdi