fivdi/pigpio #22: Change default port

craigmw, 2017-06-07:
Is there a way to change the port number that is used by pigpio? The current default is 8888 and this conflicts with an application I'm working on. I know that I could change my port on my application, but I know that pigpiod has the -p parameter which allows for changing of the port that this uses. Unfortunately, it is not possible to run pigpiod and pigpio from within node.js without a conflict.

Thanks...

Craig

fivdi, 2017-06-07:
The pigpio C library has a function called [gpioCfgSocketPort](http://abyz.co.uk/rpi/pigpio/cif.html#gpioCfgSocketPort) which can be used to configure the port number. However, the pigpio Node.js package currently doesn't allow this function to be used but it could be extended to allow it to be used.

What exactly is the goal? Would you like to run pigpiod and a Node.js process that uses pigpio at the same time? If so, I don't think this is going to be possible even if the port number is changed.

craigmw, 2017-06-07:
Brian:

The point is that I have other socket communications that need to access port 8888 on the RPi, so being able to set pigpio to access another port (say 8889) would be useful. I don't plan to use pigpiod, just to run pigpio through my node application.

Thanks...

Craig

fivdi, 2017-06-07:
Ok. I'll add a new method called `configureSocketPort` which will be similar to [configureClock](https://github.com/fivdi/pigpio/blob/master/doc/configuration.md#configureclockmicroseconds-peripheral). It'll take a few days before I get around to doing it.

craigmw, 2017-06-07:
Great! I think this will be really helpful. Thanks for this and for a really nice gpio library for node.

Craig

fivdi, 2017-06-10:
pigpio v0.6.0 which was just published on npm has a new configuration called [configureSocketPort](https://github.com/fivdi/pigpio/blob/master/doc/configuration.md#configuresocketportport) which can be used to configures socket port used by pigpio.

Please give it a try and report you findings.

craigmw, 2017-06-10:
Brian:

This works well. Even with my app running on port 8888, switching pigpio's port to 8889 works. This used to throw an error. Thanks for the quick fix!

Craig

closed 2017-06-10 by craigmw

fivdi, 2017-06-10:
That's good news. Thanks for verifying that it works.