fivdi/pigpio #23: Asynchronous callbacks

Akashbudhai, 2017-06-19:
I'm writing the code for single pin using pigpio. In the trigger pulse function will need a callback handler as parameter which can be invoked to set the pin as input for listening the echo.
https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#triggerpulselen-level

Can you make an async version of the trigger function?

Similarly it might be good to have async version of read and write operations

fivdi, 2017-06-19:
>I'm writing the code for single pin using pigpio. In the trigger pulse function will need a callback handler as parameter which can be invoked to set the pin as input for listening the echo.
https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#triggerpulselen-level

The trigger function is a synchronous blocking call that will block for 10 microseconds in this case (which is a very short time.) A callback is therefore not needed. Making this asynchronous will have a negative impact on performance. It's not something that I want to do.

> Similarly it might be good to have async version of read and write operations

This doesn't make sense as the read and write operations are non-blocking synchronous calls that only take about 500 nanoseconds to execute on a Raspberray Pi 3. Making them asynchronous would be counterproductive. All they do is set a bit in a register or read a bit in a register and they do this very quickly. They shouldn't be asynchronous.

closed 2017-06-19 by fivdi

Akashbudhai, 2017-06-22:
If read write takes about 500 ns it makes sense to not make them asynchronous like you said. However trigger pulse of 10 microseconds would be considerable time when you are interfacing with hardware. What are your thoughts about this?

fivdi, 2017-06-22:
> However trigger pulse of 10 microseconds would be considerable time when you are interfacing with hardware. What are your thoughts about this?

I disagree and my thoughts on the subject haven't changed.

On a Raspberry Pi 3 the following line of code takes about 5 microseconds to execute:
```js
  Buffer.from("Hello, World");
```
The following takes 10 microseconds to execute:
```js
  pin.trigger(10, 1);
```
So should the creation of a Buffer object also be asynchronous because it takes 5 microseconds to execute? I think not.

It doesn't make sense for trigger to be asynchronous. This would result in a ton of context switching between threads which will make everything much slower and simply burn CPU cycles for no reason. It will mean less processing time for the JavaScript thread in the Node.js process, not more.