fivdi/pigpio #25: How to enable low-level DEBUG messaging?

mwittig, 2017-06-23:
The pigpio library provides a logging facility: Is it possible to enable this at runtime? 
Perhaps it requires some helper function to set the debug level at running, e.g. configureDebugLevel()

fivdi, 2017-06-23:
The pigpio C library has a function called [gpioCfgSetInternals](http://abyz.co.uk/rpi/pigpio/cif.html#gpioCfgSetInternals) which can be used for this. Currently the pigpio Node.js package doesn't provide access to this function but it could.

It is however possible to call this function form the command line!

Start your Node.js application in one terminal.

Then in a second terminal call [pigs csi](http://abyz.co.uk/rpi/pigpio/pigs.html#CSI):
```
pigs csi 4294967295
```
Your Node.js application should then start logging information in it's terminal.

Note that pigpio doesn't document what numbers can be passed to `pigs csi` but to the best of my knowledge 4294967295 (0xffffffff) enables as much as possible but i could be mistaken here.

mwittig, 2017-06-25:
Thanks a lot! Works for me!

closed 2017-06-25 by mwittig