fivdi/pigpio #27: Calling PIGPIO from multiple task failed

bahamut657, 2017-07-08:
I have two chid processes, each one control specific PIN through pigpio.
when the second subtask starts, i obtain this error 
2017-07-08 15:02:10 initInitialise: Can't lock /var/run/pigpio.pid
~/node_modules/pigpio/pigpio.js:11
    pigpio.gpioInitialise();
           ^

Error: pigpio error -1 in gpioInitialise
    at initializePigpio (~/node_modules/pigpio/pigpio.js:11:12)
    at new Gpio (~/node_modules/pigpio/pigpio.js:25:3)

Could be possible to use pigpio from multiple task at the moment?
Why this limitation? I miss something?

fivdi, 2017-07-08:
I'm afraid that this is a limitation of the [pigpio C library](https://github.com/joan2937/pigpio) which is out of my control.

fivdi, 2017-07-08:
This should be documented in the readme.

fivdi, 2017-07-10:
The readme now has a [limitations section](https://github.com/fivdi/pigpio#limitations) which documents that the pigpio Node.js package can only be used by a single running process. This is a limitation of the pigpio C library.

@bahamut657 Thank you for reporting this issue.

closed 2017-07-10 by fivdi