fivdi/pigpio #28: Module only accessible when run with sudo

shekit, 2017-07-17:
Hey,

I am only able to run the node script with pigpio if I call it with sudo.

If I don't call it with sudo, I get 
```

Sorry, you don't have permission to run this program. Try running as root

Error: pigpio error -1 in gpioInitialise
```

fivdi, 2017-07-18:
> I am only able to run the node script with pigpio if I call it with sudo.

Yes, this is the case.
The pigpio Node.js package is a wrapper for the [pigpio C library](https://github.com/joan2937/pigpio). The pigpio C libray requires root/sudo privileges to access certain hardware peripherals. The pigpio Node.js package can't be modified to avoid this.

closed 2017-07-18 by fivdi

fivdi, 2017-07-18:
There's an alternative Node.js package called [onoff](https://github.com/fivdi/onoff) that doesn't provide as much functionality as pigpio but it doesn't require root/sudo privileges.

shekit, 2017-07-18:
Thanks, I had already taken a look at that but needed some of the functionality of pigpio. No worries, will work with sudo! Thanks!

fivdi, 2017-07-23:
The necessity for root/sudo privileges is now documented in the [Limitations section](https://github.com/fivdi/pigpio#limitations) or the readme.