fivdi/pigpio #31: PIGPIO on PI3

scargill, 2017-08-09:
Hi

I just did an absolutely normal installation of Node-Red and PIGPIO....

When I ran Node-Red (as normal Pi user) - I tried to use PIGPIO and it failed - so I ran Node-Red in the console and tried again after erasing the FLOW file so as to start from scratch. 

This is the full Raspbian 17/05/2017 installation.

Here is the part of my script normally used - and used in this case - when installing Node-Red

           if [[ $OPSYS == *"RASPBIAN"* ]]; then
                sudo sed -i -e 's#exit 0#chmod 777 /dev/ttyAMA0\nexit 0#g' /etc/rc.local
                sudo apt-get -y install python{,3}-rpi.gpio 2>&1 | tee -a $LOGFILE
                npm $NQUIET install node-red-contrib-gpio 2>&1 | tee -a $LOGFILE
                npm $NQUIET install raspi-io 2>&1 | tee -a $LOGFILE
            fi

And here is the error from PIGPIO

9 Aug 13:41:29 - [info] Starting flows
1502286089305 Available RaspberryPi-IO
9 Aug 13:41:29 - [info] Started flows
2017-08-09 13:41:29 initCheckPermitted:
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

9 Aug 13:41:29 - [red] Uncaught Exception:
9 Aug 13:41:29 - Error: pigpio error -1 in gpioInitialise
    at Error (native)
    at initializePigpio (/home/pi/.node-red/node_modules/pigpio/pigpio.js:11:12)
    at new Gpio (/home/pi/.node-red/node_modules/pigpio/pigpio.js:25:3)
    at new DigitalOutput (/home/pi/.node-red/node_modules/raspi-gpio/dist/index.js:79:24)
    at RaspiIOCore.value (/home/pi/.node-red/node_modules/raspi-io-core/dist/index.js:553:38)
    at RaspiIOCore.pinMode (/home/pi/.node-red/node_modules/raspi-io-core/dist/index.js:505:21)
    at /home/pi/.node-red/node_modules/raspi-io-core/dist/index.js:427:17
    at Array.forEach (native)
    at /home/pi/.node-red/node_modules/raspi-io-core/dist/index.js:356:32
    at _combinedTickCallback (internal/process/next_tick.js:73:7)
    at process._tickDomainCallback (internal/process/next_tick.js:128:9)
pi@pi3:~ $ ^C

fivdi, 2017-08-09:
As shown in the output:
```
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program. |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+
```
and as mentioned in the [limitations](https://github.com/fivdi/pigpio#limitations) section of the readme the pigpio C library and therefore the pigpio Node.js package requires root/sudo privileges to access hardware peripherals.

scargill, 2017-08-09:
Ok, I do understand that but then why would we have a node for Node-Red-  which is normally run as Pi user?   and if I remember rightly it is possible to run a daemon so that PIGPIO doesn't need root access... isn't that right??

scargill, 2017-08-09:
or are you saying that in the INSTALL I should have used:

SUDO npm $NQUIET install node-red-contrib-gpio 2>&1 | tee -a $LOGFILE

fivdi, 2017-08-09:
> Ok, I do understand that but then why would we have a node for Node-Red- which is normally run as Pi user?

I don't understand the question being asked here.

> and if I remember rightly it is possible to run a daemon so that PIGPIO doesn't need root access... isn't that right??

The pigpio Node.js package uses the pigpio C library directly and the pigpio C library requires root/sudo privileges. The pigpio Node.js package does not use the pigpio deamon. 

> or are you saying that in the INSTALL I should have used:

> SUDO npm $NQUIET install node-red-contrib-gpio 2>&1 | tee -a $LOGFILE

No, this is not what I'm saying.

scargill, 2017-08-10:
Ok Brian thanks for that. Basically I simply don't understand, when we're encouraged even by the guys who developed Node-Red to use user Pi and not root for general use, this needs root. Is there not something we can install that gives Pi user the right permissions?

boneskull, 2017-08-10:
I think I can clear up some confusion...

@scargill Looks like you're using `node-red-contrib-gpio`, which by way of many dependencies, uses this module.  

`node-red-contrib-gpio` supports whatever boards [Johnny-Five](/rwaldron/johnny-five) supports through IO plugins.  The IO plugin used to talk to the RPi uses *this* module, which in turn uses [the pigpio C lib](/joan2937/pigpio), which requires `root` access to run.

So, practically speaking, you would need to start Node-RED as `root` if you want to use `node-red-contrib-gpio` on a RPi.

scargill, 2017-08-10:
Hi there

Thanks for that - well for myself and many - starting as root is not practical...

For reference, I found THIS code which was not originally meant for the Pi and which needs someone to support it - but it works -  as NON-root - thanks to some mods to general permissions.  I'm guessing it is possible that one could glean info from this to allow node-red-contrib-gpio to work for pi users - which surely has to be the vast majority of users.  See the mods necessary for permissions.... less than half way down the blog entry we talk about permissions...

Would be great if this could run for Pi users with a simple mod - which I would then include in "the script" to make life easier for folk... https://tech.scargill.net/a-new-gpio-in-town/

fivdi, 2017-08-12:
I'm going to close this issue as there is no TODO for the pigpio Node.js module.

In order to make things like PWM and servo pulses possible on any pin the pigpio C library needs to access hardware peripherals on the Pi. `/dev/mem` is used to access the required hardware peripherals. root privileges are needed to access `/dev/mem`.

closed 2017-08-12 by fivdi

hibiyasleep, 2018-03-07:
couldn't we use `/dev/gpiomem`? I'm running my program with `gpio` group member, so another program's GPIO interaction (example via `raspi-gpio`) fully works without root.

fivdi, 2018-03-07:
> couldn't we use `/dev/gpiomem`?

Unfortunately not. While `/dev/gpiomem` does allow a number of peripherals to be accessed without root privileges, for example, the GPIO peripheral, it doesn't allow all peripherals to be accessed without root privileges. The pigpio C library enables PWM on any pin. To achieve this it uses DMA. `/dev/gpiomem` doesn't allow access to the DMA peripheral without root privileges.

> I'm running my program with gpio group member, so another program's GPIO interaction (example via raspi-gpio) fully works without root.

Being a member of the gpio group doesn't give the user enough privileges to access the DMA peripheral but access to the DMA peripheral is required by pigpio to make PWM on any pin possible.

`raspi-gpio` depends on pigpio and therefore requires root privileges. It's unclear to me how you managed to get `raspi-gpio` to function without root privileges.

If I run the following `raspi-gpio` program without root privileges:
```js
const raspi = require('raspi');
const gpio = require('raspi-gpio');

raspi.init(() => {
  const output = new gpio.DigitalOutput('P1-11');
  output.write(1);
  setTimeout(() => {
    output.write(0);
  }, 1000);
});
```
this is the output I see:
```
2018-03-07 10:26:40 initCheckPermitted: 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

/home/pi/raspi-gpio/node_modules/pigpio/pigpio.js:11
    pigpio.gpioInitialise();
           ^

Error: pigpio error -1 in gpioInitialise
    at initializePigpio (/home/pi/raspi-gpio/node_modules/pigpio/pigpio.js:11:12)
    at new Gpio (/home/pi/raspi-gpio/node_modules/pigpio/pigpio.js:25:3)
    at new DigitalOutput (/home/pi/raspi-gpio/node_modules/raspi-gpio/dist/index.js:79:25)
    at raspi.init (/home/pi/raspi-gpio/test.js:5:18)
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
    at process._tickCallback (internal/process/next_tick.js:180:9)
    at Function.Module.runMain (module.js:686:11)
    at startup (bootstrap_node.js:187:16)
    at bootstrap_node.js:608:3
```

There are however modules that allow access to GPIOs without root privileges. An example is [onoff](https://github.com/fivdi/onoff).

scargill, 2018-03-07:
Indeed. It is common practice NOT to use root. Nofr-Red for Eample would normally be used as user PI, so something that needs root access is not helpful