fivdi/pigpio #33: .h

rcomeau, 2017-08-19:
pigpio.h missing

fivdi, 2017-08-20:
Was there an error message displayed while installing the pigpio Node.js package saying that pigpio.h is missing? If so, it's probably because the pigpio C library is not installed. Newer versions of Raspbian Jessie come with the pigpio C library installed. Raspbian Jessie Lite and older versions of Raspbian don't come with the pigpio C library installed and the pigpio C library needs to be installed manually with the following commands:
```
sudo apt-get update
sudo apt-get install pigpio
```
If the pigpio C library is installed it should then be possible to install the pigpio Node.js package.

Further information can be found in the [Installation Section](https://github.com/fivdi/pigpio#installation) of the readme.

If this doesn't resolve the issue please provide further information about the error and how it can be reproduced.

rcomeau, 2017-08-22:
I was working with two Raspberry Pis, and one somehow didn't get loaded with pigpio. My bad, all works fine when you load the C lib.
Rick Comeau

closed 2017-08-22 by fivdi