fivdi/pigpio #34: pigpio.initialize() doenst do its job after pigpio.terminate()

joshi11, 2017-09-18:
I have a situation in which I use pigpio in Node and in native C
Thus I need to call pigpio.terminate() in Node before I can call pigpio in C otherwise C complains about not getting the lock on the pid.
When I then try to call pigpio.initialize in Node again it doesnt seem to work since I get the message:

> calling initialize
> Hardware Revision: 9000c1
> initialized= 1
> 2017-09-17 18:45:59 gpioSetMode: pigpio uninitialised, call gpioInitialise()

```

var initialized;
var pigpio = require('pigpio')
var Gpio=pigpio.Gpio

 exports.test = function(led_nr, dutyCycle) {	
	if (!initialized)	{
		console.log("calling initialize");
		pigpio.initialize(); // pigpio C library initialized here
		initialized=1;
		console.log('Hardware Revision: ' + pigpio.hardwareRevision().toString(16));
		test_led(led_nr, dutyCycle);   
	}
	else 
		test_led(led_nr, dutyCycle);    //called function to make sure I dont get race conditions
	
	
};	

exports.terminate = function() {
	console.log("initialized= " + initialized);
	if (initialized)	{
		pigpio.terminate(); // pigpio C library terminated here
		initialized=0;
		console.log("initialized= " + initialized);
		console.log('Terminated');
	}	//end if
};//end terminate

    function test_led(led_nr, dutyCycle)	{
	console.log("initialized= " + initialized);
	led = Gpio(led_nr, {mode: Gpio.OUTPUT}); 
	
	led.pwmRange(100)
	led.pwmWrite(dutyCycle);
}
```

Versions:

> pigpio: "_id": "pigpio@0.6.1",
> node v6.11.3

I expirience no problems with the C version of the pigpio library - hence I would assume its an issue with the wrapper.

fivdi, 2017-09-18:
Does it work if the [following code in pigpio.js](https://github.com/fivdi/pigpio/blob/master/pigpio.js#L303-L305) is modified from:
```js
module.exports.terminate = function () {
  pigpio.gpioTerminate();
};
```
to:
```js
module.exports.terminate = function () {
  pigpio.gpioTerminate();
  initialized = false;
};
```

joshi11, 2017-09-18:
Yes!!! 
Thank you!

fivdi, 2017-09-18:
Ok, thanks for the feedback.
I'll release a new version on npm with a fix as soon as possible.

fivdi, 2017-09-18:
Fixed by 1f745d2c4ee06f95ab38e764ab4a138c97044eda in pigpio@0.6.2.

closed 2017-09-18 by fivdi