fivdi/pigpio #35: Accessing the OK LED

wordyallen, 2017-09-26:
Hello, 

Before I installed the C library I was able to access the onboard OK LED.  Using something like
`fs.writeFileSync('/sys/class/leds/led0/brightness', '1')`

However, now it turns on and then gets overwritten.. Is there a way to control this with your library?
I tried GPIO 16  and 47(https://www.raspberrypi.org/forums/viewtopic.php?f=72&t=114796) . I couldnt get it. 

Thanks!

wordyallen, 2017-09-26:
Oh.. I see it overwrote the `/sys/class/leds/led0/trigger` file.

fivdi, 2017-09-26:
If the string "none" is written to `/sys/class/leds/led0/trigger` it should then be possible to turn the LED on or off by writing to `/sys/class/leds/led0/brightness`.

fivdi, 2017-09-26:
Writing to GPIO 47 only works on the Raspberry Pi 2. This no longer works on the Raspberry Pi 3. Are you using a Raspberry Pi 3?

wordyallen, 2017-09-26:
I'm using the [Pi 2 Model B]( http://a.co/8LD4Nbb) . 

Here's my code and output:
```js
var Gpio = require('pigpio').Gpio,
  led = new Gpio(47, {mode: Gpio.OUTPUT}),
  dutyCycle = 0;

setInterval(function () {
  led.pwmWrite(dutyCycle);

  dutyCycle += 5;
  if (dutyCycle > 255) {
    dutyCycle = 0;
  }
}, 20);
```
`2017-09-26 22:34:31 gpioPWM: bad gpio (47)`

wordyallen, 2017-09-26:
ahh.. it the pwm on a non pwm pin... sorry,. I got it working now!

closed 2017-09-27 by wordyallen