fivdi/pigpio #36: pigpio.cc:2:10: fatal error: 'pigpio.h' file not found

iroy2000, 2017-10-08:
* **Node Version**: 

`npm -v`
`5.4.1`

`node -v`
`v8.4.0`

* **Platform**: 

`macOS . 10.12.6`

* **Compiler**: 
```
Apple LLVM version 9.0.0 (clang-900.0.37)
Target: x86_64-apple-darwin16.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

* **Module**: 
https://github.com/marak/say.js/

<details><summary>Verbose output (from npm or node-gyp):</summary>

```
> fibers@1.0.15 install /Users/roy/development/tjbot-code-challenge/node_modules/fibers
> node build.js || nodejs build.js

  CXX(target) Release/obj.target/fibers/src/fibers.o
../src/fibers.cc:122:18: warning: 'NewFromOneByte' is deprecated [-Wdeprecated-declarations]
                return String::NewFromOneByte(isolate, (const uint8_t*)string);
                               ^
/Users/roy/.node-gyp/8.4.0/include/node/v8.h:2624:10: note: 'NewFromOneByte' has been explicitly marked deprecated here
  static V8_DEPRECATED(
         ^
/Users/roy/.node-gyp/8.4.0/include/node/v8config.h:332:29: note: expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated))
                            ^
../src/fibers.cc:126:18: warning: 'NewFromOneByte' is deprecated [-Wdeprecated-declarations]
                return String::NewFromOneByte(isolate, (const uint8_t*)string);
                               ^
/Users/roy/.node-gyp/8.4.0/include/node/v8.h:2624:10: note: 'NewFromOneByte' has been explicitly marked deprecated here
  static V8_DEPRECATED(
         ^
/Users/roy/.node-gyp/8.4.0/include/node/v8config.h:332:29: note: expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated))
                            ^
../src/fibers.cc:433:80: warning: 'NewInstance' is deprecated [-Wdeprecated-declarations]
                                return uni::Return(uni::Deref(Isolate::GetCurrent(), tmpl)->GetFunction()->NewInstance(1, argv), args);
                                                                                                           ^
/Users/roy/.node-gyp/8.4.0/include/node/v8.h:3787:3: note: 'NewInstance' has been explicitly marked deprecated here
  V8_DEPRECATED("Use maybe version",
  ^
/Users/roy/.node-gyp/8.4.0/include/node/v8config.h:332:29: note: expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated))
                            ^
../src/fibers.cc:621:14: warning: 'TryCatch' is deprecated [-Wdeprecated-declarations]
                                TryCatch try_catch;
                                         ^
/Users/roy/.node-gyp/8.4.0/include/node/v8.h:8157:3: note: 'TryCatch' has been explicitly marked deprecated here
  V8_DEPRECATED("Use isolate version", TryCatch());
  ^
/Users/roy/.node-gyp/8.4.0/include/node/v8config.h:332:29: note: expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated))
                            ^
../src/fibers.cc:741:34: warning: 'ToNumber' is deprecated [-Wdeprecated-declarations]
                        Coroutine::pool_size = value->ToNumber()->Value();
                                                      ^
/Users/roy/.node-gyp/8.4.0/include/node/v8.h:2279:10: note: 'ToNumber' has been explicitly marked deprecated here
  inline V8_DEPRECATED("Use maybe version", Local<Number> ToNumber() const);
         ^
/Users/roy/.node-gyp/8.4.0/include/node/v8config.h:332:29: note: expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated))
                            ^
../src/fibers.cc:793:8: warning: 'SetAccessor' is deprecated [-Wdeprecated-declarations]
                        fn->SetAccessor(uni::NewLatin1Symbol(isolate, "current"), GetCurrent);
                            ^
/Users/roy/.node-gyp/8.4.0/include/node/v8.h:3120:3: note: 'SetAccessor' has been explicitly marked deprecated here
  V8_DEPRECATED("Use maybe version",
  ^
/Users/roy/.node-gyp/8.4.0/include/node/v8config.h:332:29: note: expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated))
                            ^
../src/fibers.cc:794:8: warning: 'SetAccessor' is deprecated [-Wdeprecated-declarations]
                        fn->SetAccessor(uni::NewLatin1Symbol(isolate, "poolSize"), GetPoolSize, SetPoolSize);
                            ^
/Users/roy/.node-gyp/8.4.0/include/node/v8.h:3120:3: note: 'SetAccessor' has been explicitly marked deprecated here
  V8_DEPRECATED("Use maybe version",
  ^
/Users/roy/.node-gyp/8.4.0/include/node/v8config.h:332:29: note: expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated))
                            ^
../src/fibers.cc:795:8: warning: 'SetAccessor' is deprecated [-Wdeprecated-declarations]
                        fn->SetAccessor(uni::NewLatin1Symbol(isolate, "fibersCreated"), GetFibersCreated);
                            ^
/Users/roy/.node-gyp/8.4.0/include/node/v8.h:3120:3: note: 'SetAccessor' has been explicitly marked deprecated here
  V8_DEPRECATED("Use maybe version",
  ^
/Users/roy/.node-gyp/8.4.0/include/node/v8config.h:332:29: note: expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated))
                            ^
8 warnings generated.
  CXX(target) Release/obj.target/fibers/src/coroutine.o
  CC(target) Release/obj.target/fibers/src/libcoro/coro.o
  SOLINK_MODULE(target) Release/fibers.node
Installed in `/Users/roy/development/tjbot-code-challenge/node_modules/fibers/bin/darwin-x64-57/fibers.node`

> pigpio@0.4.0 install /Users/roy/development/tjbot-code-challenge/node_modules/pigpio
> node-gyp rebuild

  CXX(target) Release/obj.target/pigpio/src/pigpio.o
../src/pigpio.cc:2:10: fatal error: 'pigpio.h' file not found
#include <pigpio.h>
         ^~~~~~~~~~
1 error generated.
make: *** [Release/obj.target/pigpio/src/pigpio.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:125:13)
gyp ERR! stack     at ChildProcess.emit (events.js:213:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Darwin 16.7.0
gyp ERR! command "/usr/local/Cellar/node/8.4.0/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/roy/development/tjbot-code-challenge/node_modules/pigpio
gyp ERR! node -v v8.4.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
```

```

```

</details>

fivdi, 2017-10-10:
Currently the pigpio module can only be installed on Raspbian.

fivdi, 2017-10-10:
pigpio.h can't be found because the pigpio C library is not installed. See step 1 of the installation instructions https://github.com/fivdi/pigpio#step-1---install-the-pigpio-c-library

fivdi, 2017-10-15:
I'm going to go ahead and close this issue. There isn't an issue with pigpio. If this turns out not to be the case the issue can be reopened.

closed 2017-10-15 by fivdi

jdoleczek, 2019-08-22:
The problem still exists :-(

$ npm i -S pigpio

> pigpio@1.2.3 install /home/pi/third/node_modules/pigpio
> node-gyp rebuild

make: Entering directory '/home/pi/third/node_modules/pigpio/build'
  CXX(target) Release/obj.target/pigpio/src/pigpio.o
../src/pigpio.cc:2:10: fatal error: pigpio.h: No such file or directory
 #include <pigpio.h>
          ^~~~~~~~~~
compilation terminated.
make: *** [pigpio.target.mk:103: Release/obj.target/pigpio/src/pigpio.o] Error 1
make: Leaving directory '/home/pi/third/node_modules/pigpio/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:196:23)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 4.19.66-v7+
gyp ERR! command "/usr/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/third/node_modules/pigpio
gyp ERR! node -v v10.15.2
gyp ERR! node-gyp -v v5.0.3
gyp ERR! not ok 
npm WARN third@1.0.0 No description
npm WARN third@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! pigpio@1.2.3 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the pigpio@1.2.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

fivdi, 2019-08-22:
Is the pigpio C library installed on your Raspberry Pi? Note that I mean the pigpio C library here and not the pigpio Node.js module.

If the pigpio C library is installed, running the following command will display the version on the pigpio C library that is installed:
```
pigpiod -v
```
If the pigpio C library is not installed, it will need to be installed before installing the pigpio Node.js module. For further details see the [Installation](https://github.com/fivdi/pigpio#installation) section of the readme.