fivdi/pigpio #37: pigpio on docker image

rafaelquines, 2017-10-10:
Hello.

I'd like use your lib in a docker container inside raspberry pi.

Can you tell me what volumes I need map to container access rasp gpios?

Creating volumes, it works transparent for application.

Thanks

fivdi, 2017-10-10:
I'm afraid I'm not familiar enough with docker to answer this question.

rafaelquines, 2017-10-10:
Ok, thanks, but what filesystem this lib use to manipulate gpios?

Example: /sys/class/gpio, /dev/...

Do you know? It could be solve my doubt.

fivdi, 2017-10-13:
The pigpio Node.js module doesn't access any files directly. It calls functions from the pigpio C library (https://github.com/joan2937/pigpio) to perform all operations. I'm not familiar enough with the internal implementation details of the C library to list all files it uses. Among other things it uses /dev/mem and /sys/class/gpio. However, I'm fairly certain it uses other files too.

fivdi, 2017-10-22:
For more information it's probably best to ask over at https://github.com/joan2937/pigpio/issues. I don't think there is anything actionable for the pigpio Node.js module here so I'm going to go ahead and close the issue.

closed 2017-10-22 by fivdi