fivdi/pigpio #38: Waveforms

hsmptg, 2017-10-12:
Great pigpio Node support but I miss the waveforms that the C and Python implementations allow.
Without them I do not see any easy way of generating serial signals.
I hope to see this addition soon.
Regards

itpmngt, 2018-01-01:
plus +1 to this feature - it will bring PISCOPE to node, awesome...

Great set of node modules for the PI - Thank You!!!

AnthonyMyatt, 2018-02-04:
For those interested I have implemented most of these functions as I needed them to generate a dynamic IR signal to control an Air Conditioner unit via Apple's HomeKit.

I will continue to implement the rest of the functions and create a pull request once done.

[https://github.com/AnthonyMyatt/pigpio/tree/Waves](https://github.com/AnthonyMyatt/pigpio/tree/Waves)

fivdi, 2018-02-05:
@AnthonyMyatt This is fantastic. A pull request would be more than welcome.

depuits, 2018-07-16:
@AnthonyMyatt are you still planning to create a pull request for this issue?

AnthonyMyatt, 2018-07-24:
Hi @depuits, Yes I do still plan on creating a pull request. Sorry I have been busy with work and hobby projects have been put on the back burner.

clowrey, 2018-08-27:
Would be awesome to have Software Serial function in Node-Red through this :)

bahamut657, 2018-09-01:
Awesome work done for waveforms support. Is possible to add support to function `wave_chain`?

bahamut657, 2018-09-02:
It is not possible to compile pigpio.cc using the Waves branch @AnthonyMyatt . Can you help me resolving? I can contribute to the project if you need...
I'm trying to build with node-gyp using NodeJS LTS 8.11.4 or the current 10.9.0 on RaspberryPI 3 but it results in build errors.See attachment. 
[pigpio_waveform_build.log](https://github.com/fivdi/pigpio/files/2343337/pigpio_waveform_build.log)

fivdi, 2018-09-02:
Perhaps the @AnthonyMyatt Waves branch wasn't tested with Node.js 8 or 10 yet and currently only functions with older versions of Node.js.

bahamut657, 2018-09-02:
Yes, I see...I can try to port his work on your source (your is compiling with all NodeJS versions, the other one doesn't but implements [partially] the missing functionality). Is it a good choice? Some suggestions to give to me?

AnthonyMyatt, 2018-09-03:
Hi @bahamut657, I have merged fivdi's master branch in with my branch in an attempt to bring it up to date and therefore compile with Node.js 8/10. I am away from any Raspberry-Pi devices at the moment so I am unable to test. You are welcome to give it a go. If it does compile and run could you please let me know.

bahamut657, 2018-09-03:
Wow!!! This is a wonderful news!! I will try to build at lunch time and make you know. Thanks for the fast support @fivdi  and @AnthonyMyatt

bahamut657, 2018-09-03:
@AnthonyMyatt  I try to build after updating sources, but there are some build errors.
Show the log attached
[pigpio_wave_merged_build.log](https://github.com/fivdi/pigpio/files/2345337/pigpio_wave_merged_build.log)

Thanks in advance

AnthonyMyatt, 2018-09-04:
Hi @bahamut657, my apologies. Apparently I never pushed the working code to GitHub back in February. I just logged onto my deployed Pi to find it had more recent code changes.
I have now pushed an up to date and working copy to my Waves branch.

Please let me know if you still have problems.

bahamut657, 2018-09-04:
Very very good news @AnthonyMyatt . It builds with no problem. 
So this evening I will give a test to it.
@fivdi, in case that this source is working, do you think we can merge it into Master branch?

bahamut657, 2018-09-04:
I can propose myself if some work is needed and if @AnthonyMyatt has no time to do it.
If you like I can contribute with my knowledgment. I'm a JavaScript expert but cannot help in C++ side (and v8 that,once looked at the code, is a pain).
Let me know, happy to help this project. 
PIGPIO,ported to NodeJS  is the best option in performance and functionality

bahamut657, 2018-09-04:
@AnthonyMyatt  this evening I test deeply your implementation.
All seems to work good and I add the support to `gpioWaveChain` and `gpioTick`
Files modified in attachment.

[pigpio.cc.gz](https://github.com/fivdi/pigpio/files/2350345/pigpio.cc.gz)
[pigpio.js.gz](https://github.com/fivdi/pigpio/files/2350349/pigpio.js.gz)

When you have validated and merged this work, I want to speak about performance difference between 
 this module and the pigpio.py (using the Wave functionality) 
I've ported the native _433D.c to NodeJS using this module, but the signal received are less the then C implementation.
Same thing if I compare _433D.py to this module, Python receives all the messages, NodeJS receive good message 4 times on 10.
Let me know 
Regards
PaPo

footniko, 2019-05-14:
Any updates on this?

yboujraf, 2020-01-06:
Dear @fivdi 

I can t see wave in the repository.

Am i wrong ?

Best regards 
Youssef

fivdi, 2020-01-06:
@yboujraf No, you're not wrong. Support for waveforms hasn't been added to the pigpio Node.js module yet. The most recent work on the topic can be found in the Waves branch of @Rogerrrrrrrs pigpio fork at https://github.com/Rogerrrrrrrs/pigpio/tree/Waves

What is your use case for waveforms?

yboujraf, 2020-01-06:
Dear @fivdi 

Just seen the message, too busy today because I have some questions about PIGPIO with my raspberry PI4B but will back soon for the use cases.

Best Regards,
Youssef

PR #95 (2020-01-11): Adds support for waves functions
This pull request adds Wave functionality and resolves #38 and closes #64 and is basically a continuation of #64  

It also adds ticks to the interrupt event as discussed [here](https://github.com/fivdi/pigpio/pull/64#issuecomment-433185029)

The following methods where added:
* waveClear
* waveAddNew
* waveAddGeneric
* waveCreate
* waveDelete
* waveTxSend
* waveChain
* waveTxAt
* waveTxBusy
* waveTxStop
* waveGetMicros
* waveGetHighMicros
* waveGetMaxMicros
* waveGetPulses
* waveGetHighPulses
* waveGetMaxPulses
* waveGetCbs
* waveGetHighCbs
* waveGetMaxCbs

Th [... cut]
closed 2020-01-11 by fivdi