fivdi/pigpio #39: There is no debounce to use push buttons

rolfjunior, 2017-11-03:
Pleas add a debounce function or code to make push buttons work with node.js.
I'm getting interference from lights and refrigerator that my raspberry assume is a button pushed.

fivdi, 2017-11-03:
>  I'm getting interference from lights and refrigerator that my raspberry assume is a button pushed.

If your lights or refrigerator are causing enough interference to be detected as a button press then I don't see how debouncing will help. A signal is debounced so that a single press doesn't appear like multiple presses. In your case the signal from the lights or refrigerator is being detected so the assumption is that the button was pressed. Even if this signal was debounced it would still be detected as a button press.

Does your button circuit have an external pull down resistor as shown [here](https://github.com/fivdi/pigpio#usage)? The 10k resistor is the pull down resistor. The Raspberry Pi does have internal pull down resistors for GPIOs but they are very weak and not really suitable for buttons.

rolfjunior, 2017-11-03:
> Does your button circuit have an external pull down resistor as shown here? The 10k resistor is the pull down resistor. The Raspberry Pi does have internal pull down resistors for GPIOs but they are very weak and not really suitable for buttons.

I was using a 10k resistor, I changed it for a 220ohm resistor and now it seems to work without interference.
Thank you!

fivdi, 2017-11-03:
No problem :)
Good to hear that it's working now.

I'll go ahead and close the issue.

closed 2017-11-03 by fivdi