fivdi/pigpio #41: Successive crashes/restarts makes alerts unresposive.

baslking, 2017-12-23:
I've read all I can find here, and everything works as expected, the Alert event with the microsecond tick is great, but I have on nagging issue:

During tests, with ctrl-c or crashes, (not a repeatable number or precise situation) the Alerts stop being generated and the only fix that I have found is to reboot my Pi.  I am not using any special signal handling routines.  I'm only watching one pin while writing to several others.  I seem to be able to continue reading and writing states, but no interrupts/alerts.   Is this just me?

I'm on node 8.9.3 LTS and  A 4.9.70-v7+ Kernel on a Pi-3 B+ with a  1.60 version of the  pigpio C library.

fivdi, 2017-12-24:
With the following software installed:

* Raspbain Debian: 9.1 (cat /etc/debian_version)
* Kernel: 4.9.59-v7+ (uname -a)
* pigpio C library: V64 (cat /usr/include/pigpio.h | grep VERSION)
* pigpio C library: 1.64-1 (apt-cache show pigpio)
* Node.js: v8.9.3 (node -v)
* pigpio: v0.6.3 (npm list)

The following test was run 100 times:
```js
const Gpio = require('pigpio').Gpio;
const led = new Gpio(17, {
  mode: Gpio.OUTPUT,
  alert: true
});

// ensure GPIO17 is low before we start monitoring the state of GPIO17 with alerts.
led.digitalWrite(0);

(function () {
  let count = 0;
  let startTick;

  // Use alerts to determine how long the LED was turned on
  led.on('alert', function (level, tick) {
    let endTick;
    let diff;

    if (level == 1) {
      startTick = tick;
    } else {
      endTick = tick;
      diff = (endTick >> 0) - (startTick >> 0); // Unsigned 32 bit arithmetic

      count += 1;

      if (count % 1000 === 0) {
        console.log(count + ' ' + diff);
      }

      setImmediate(function () {
        led.trigger(10, 1);
      });
    }
  });
}());

led.trigger(10, 1);
```
The test program pulses GPIO17 high for 10 microseconds as often as possible and uses alerts to determine how long GPIO17 was actually set high.

The program was run 100 times and terminated with ctrl+c on every run after a few thousand pulse widths were measured. Everything worked flawlessly and it was possible to measure the width of all pulses generated. It was not necessary to reboot the Pi at any stage.

Unfortunately I can't reproduce the error with the information provided. Can you provide a test program that can be used to reproduce the error please?

Additional thoughts
* Where are the alerts in your system coming from and are you 100% sure that they are always coming?
* The constellation in your system is almost certainly different to the test program above.
* The pigpio C library has several signal handlers that are there to terminate the pigpio C library correctly. For example, if ctrl+c is hit, there's a signal handler in the pigpio C library that handles this and ensures that the pigpio C library terminates correctly. If your program is crashing in a way that prevents the signal handlers from being executed then the pigpio C library may not be terminated correctly and a reboot may be necessary to get things to work again. Is your program preventing signal handlers in the pigpio C library from being executed?
* Does your system make any assumptions about the state of the pin being watched? For example, when the system starts does it assume that the pin is in a certain state, for example, 0, and if so is the pin really in state 0 when the system starts?

baslking, 2017-12-24:
Newer kernel  4.9.70-v7+
Older Raspian: 8.0
Older pigpio v60

I implemented the onoff package (that requires catching signals and un-registering interrupts), and it works reliably, my  requirements are met by the /proc based version, so I'll probably stick with that.  e to

Crashes were mostly due to stupidity on my part.

Thanks for the quick response

fivdi, 2017-12-24:
If you're switching over to onoff and there no sample code for reproducing the issue I guess the issue can be closed, correct?

baslking, 2017-12-28:
Sure, that's fine, I might try again later, I suspect it's a matter of correctly freeing the interrupt callbacks for all signals as I did on onoff

closed 2017-12-28 by baslking

fivdi, 2017-12-28:
Ok, thanks for the feedback.