fivdi/pigpio #42: Error -123 in gpioSetISRFunc

andrewpatterson3001, 2017-12-29:
I've created a js file that captures motion from a PIR sensor on a Raspberry Pi and and trying to test it but I get this error when I run it:
```
pi@raspberrypi:~/squirrel $ sudo "$(which node)" soundtest.js
1514575499162 Available RaspberryPi-IO
1514575499514 Connected RaspberryPi-IO
1514575499524 Repl Initialized
>> starting
/home/pi/squirrel/node_modules/pigpio/pigpio.js:137
  pigpio.gpioSetISRFunc(this.gpio, +edge, +timeout, handler);
         ^
Error: pigpio error -123 in gpioSetISRFunc
    at Gpio.enableInterrupt (/home/pi/squirrel/node_modules/pigpio/pigpio.js:137:10)
    at new DigitalInput (/home/pi/squirrel/node_modules/raspi-gpio/dist/index.js:116:22)
    at RaspiIOCore.value (/home/pi/squirrel/node_modules/raspi-io-core/dist/index.js:550:38)
    at RaspiIOCore.pinMode (/home/pi/squirrel/node_modules/raspi-io-core/dist/index.js:505:21)
    at Motion.value (/home/pi/squirrel/node_modules/johnny-five/lib/motion.js:29:17)
    at new Motion (/home/pi/squirrel/node_modules/johnny-five/lib/motion.js:185:10)
    at Board.<anonymous> (/home/pi/squirrel/soundtest.js:11:16)
    at emitNone (events.js:111:20)
    at Board.emit (events.js:208:7)
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
```

I'm not really sure what this means. The code that I am running is:
```
const player = require('play-sound')();

const raspi = require('raspi-io');
const five = require('johnny-five');
const board = new five.Board({io: new raspi()});

board.on("ready", function() {

  // Create a new `motion` hardware instance.
  console.log('starting')
  let motion = new five.Motion('P1-7');
  // "calibrated" occurs once, at the beginning of a session,
  motion.on("calibrated", function() {
    console.log("calibrated");
  });
...do things with the motion
});
```
I'm using Node 8.9.3 and pigpio 0.6.4. If you have any advice, I would greatly appreciate it. Thanks!

fivdi, 2017-12-29:
This error occurs when the pigpio C library attempts to enable interrupts for a GPIO but can't. In this case the program is attempting to enable interrupts for GPIO4 which is the GPIO broken out to pin P1-7 on the Raspberry Pi GPIO header.

On a Raspberry Pi the one-wire interface can be enabled on GPIO4. If one-wire is enabled on GPIO4 then GPIO4 is reserved for the one-wire interface and can't be used for something else. Is one-wire enabled on your Raspberry Pi? If so, GPIO4 can't be used for motion detection and a different GPIO will have to be used. Alternatively, one-wire could be disabled and GPIO4 could be used for motion detection.

If you are are unsure whether or not one-wire is enabled on your Raspberry Pi please post the complete content of file `/boot/config.txt`.

andrewpatterson3001, 2017-12-30:
That was exactly the problem. I disabled 1-wire and did not encounter the error again. Thank you very much!

closed 2017-12-30 by andrewpatterson3001

fivdi, 2017-12-30:
> That was exactly the problem. I disabled 1-wire and did not encounter the error again. Thank you very much!

You're welcome and thank you for the feedback.

Clyde-arcega, 2019-10-27:
Hi i also encountered the same problem and disabling the 1-wire is not working and the error still occurs.

fivdi, 2019-10-27:
@Clyde-arcega please open a new issue to describe the problem you are having in full detail. Please include a short program that can be used to reproduce the problem.