fivdi/pigpio #44: Glitch filtering

marcinbudny, 2018-02-21:
Is it possible to use the pigpio's [gpioGlitchFilter](http://abyz.me.uk/rpi/pigpio/cif.html#gpioGlitchFilter) with this library?

fivdi, 2018-02-21:
Currently this isn't possible.

fivdi, 2018-02-21:
It looks like it would be relatively easy to add.

marcinbudny, 2018-02-22:
This is good news. Similar to glitch filtering is the [gpioNoiseFilter](http://abyz.me.uk/rpi/pigpio/cif.html#gpioNoiseFilter). 
Unfortunately I have no experience with native bindings in node, so I will have to catch up on that before attempting a PR.

DvdBerg, 2018-05-16:
I also needed this feature, so I've decided to add it: #50

closed 2018-05-20 by fivdi

fivdi, 2018-05-20:
@DvdBerg @marcinbudny pigpio@1.1.0 was just published on npm has a new [glitchFilter method](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#glitchfiltersteady).

@DvdBerg Thank you very much to taking the time to create the pull request.

louisvangeldrop, 2019-01-17:
What about the gpioNoiseFilter? Using a RF433 receiver will cause thousands of noise interrupts per seconds killing the cpu of my Rasperry Pi.

fivdi, 2019-01-17:
As mentioned above gpioNoiseFilter is supported. Please take a look at the documentation.

louisvangeldrop, 2019-01-17:
Please point me to the document in https://github.com/fivdi/pigpio? I can't find it in the npm project

fivdi, 2019-01-17:
Sorry, my mistake, the title of this issue is "Glitch Filtering" so I assumed you were asking about glitch filtering and now see that you are asking about noise filtering. gpioNoiseFilter isn't actually supported by this Node.js module at this time. Only [glitch filtering](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#glitchfiltersteady) is supported.

Please note that the documentation for the pigpio C library functions [gpioGlitchFilter](http://abyz.me.uk/rpi/pigpio/cif.html#gpioGlitchFilter) and [gpioNoiseFilter](http://abyz.me.uk/rpi/pigpio/cif.html#gpioNoiseFilter) state that calling these functions has no affects on interrupts. Calling them does however have affects on alerts.

louisvangeldrop, 2019-01-17:
I am using the alert function.

fivdi, 2019-01-17:
Would you like to create a PR for adding gpioNoiseFilter?

louisvangeldrop, 2019-01-18:
No idea how to do that. I have no experience in C++ programming

PR #50 (2018-05-20): gpioGlitchFilter
Closes #44. Add support for the `gpioGlitchFilter` function. This PR contains the following changes:

- `gpioGlitchFilter` native binding.
- Test script
- Example script, which uses `gpioGlitchFilter` to debounce a push button (also included a Fritzing picture of the circuit).
- Updated the docs and readme.