fivdi/pigpio #45: Alert and Interrupt events

depuits, 2018-02-23:
Whats the difference between the interrupt and alert? As far as I can see the alert is the same as the interrupt but without the options, but the event does get the tick while the interrupt event doesn't receive the tick. What's the reason to not pass this tick along?

fivdi, 2018-02-24:
> Whats the difference between the interrupt and alert?

There's actually quite a big difference but it may not be immediately obvious. Take a look at the documentation for the [pigpio C library gpioSetAlertFunc](http://abyz.me.uk/rpi/pigpio/cif.html#gpioSetAlertFunc) and the [pigpio C library gpioSetISRFunc](http://abyz.me.uk/rpi/pigpio/cif.html#gpioSetISRFunc).

`gpioSetAlertFunc` is used for alerts. It provides information about state changes once per millisecond so there's quite a bit of latency. However, it provides very accurate `tick` information and can detect a lot more state changes per second than `gpioSetAlertFunc`.

`gpioSetISRFunc` is used for interrupts. It provides information about state changes as quickly as possible so there's a lot less latency than with `gpioSetAlertFunc`. However, the `tick` information it provides is not as accurate as the `tick` information provided by `gpioSetAlertFunc`. In addition, `gpioSetISRFunc` can't detect as many state changes per second as `gpioSetAlertFunc`.

> What's the reason to not pass this tick along?

The pigpio Node.js module doesn't provide `tick` information for interrupts because I decided that it's not accurate enough.

depuits, 2018-02-24:
Thanks, this answers my question. The latency is indeed described in the C library documentation but its easy to read over. Maybe this could be added to the documentation here to make the difference more clear in this library, because I thought I needed the interrupt til I couldn't find the tick in the interrupt event.

fivdi, 2018-02-24:
@depuits Thanks for the feedback. Adding a node to the documentation makes sense. I'll add one.

fivdi, 2018-03-08:
A section was added to the documentation to describe the difference between interrupts and alerts. See [here](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#the-difference-between-interrupts-and-alerts).

closed 2018-03-08 by fivdi