fivdi/pigpio #46: something throwing max listeners I swear it's not me :-)

dkebler, 2018-02-23:
using the gpio interrupt

I only set the 'interrupt' listener a single time.  Then of course when the pin changes state your code emits an 'interrupt'.

So it's a bit concerning that after a bunch of emits a max listeners is thrown as I have only added a single listener.  Are there some behind the scenes events that your code or C code makes listeners for that may either be not removed or not set to .once?  cause I can't understand how it's my code since I only set one listener before calling enable.  Emitting many times to single listener does not throw this error.

```
starting interrupt on pin 10
edge= 0
Interrupt occured for 1th time on pin 10
.
.
.
.
Interrupt occured for 110th time on pin 10
Interrupt occured for 111th time on pin 10
Interrupt occured for 112th time on pin 10
Interrupt occured for 113th time on pin 10
Interrupt occured for 114th time on pin 10
(node:2204) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 drain listeners added. Use emitter.setMaxListeners() to increase limit
```
      let pin = this.interrupts[pin_num].pin
      let edge = this.interrupts[pin_num].edge
      console.log(`starting interrupt on pin ${pin_num}`)
      pin.on('interrupt', this.interruptProcess.bind(this,pin_num))
      // rock n roll!!, start the pigpio interrupt
      console.log('edge=',edge)
      if(!this.mock) pin.enableInterrupt(edge)
```

fivdi, 2018-02-24:
The test program https://github.com/fivdi/pigpio/blob/master/test/isr-performance.js can detect in excess of 20,000 interrupts per second without issue. There are no warnings related to the maximum number of listeners. Please provide a complete and short program that can be used to reproduce the error.

dkebler, 2018-02-24:
Dang it was me, nevermind :-).  Just commented out a socket write that's part of the 'interrupt' callback and error doesn't come after several thousand.   Guess I have a bug to track down.

closed 2018-02-24 by dkebler

fivdi, 2018-02-24:
> Dang it was me, nevermind

No problem :smile: