fivdi/pigpio #49: Using internal pull up/down resistor causes constant interrupts

JasXSL, 2018-03-26:
I dunno if this is a fault in my wiring or a software issue. I have the following code:

<pre>
button= new Gpio(pin, {
        mode: Gpio.INPUT,
        pullUpDown : Gpio.PUD_DOWN,
        edge : Gpio.EITHER_EDGE
})
button.on("interrupt", function(level){
        console.log("Pin", pin, "Level", level)
})
</pre>

With my wiring being GPIO -> Button -> 3v3

The levels are proper, so it's not floating, but the interrupt function runs constantly.

However, when I make a proper pull down setup using external resistors, it works fine and only interrupts when I actually hit or release the button. Am I missing something?

This is tested on a v2 B+, and v3 with the same results.

fivdi, 2018-03-26:
It's not unusual to see effects like this. It's an electrical issue rather than a software issue. Internal pulldown resistors are weak, about 50kΩ, and not necessarily good enough to function as pulldown resistors for buttons. External pulldown resistors are typically a lot stronger, for example 10kΩ, and a lot more effective.

JasXSL, 2018-03-26:
@fivdi Yep you're right. If anyone runs into this issue, my solution was to add a small capacitor between GPIO and ground.

closed 2018-03-26 by JasXSL