fivdi/pigpio #52: Sound stream breaks when initing the library

sznowicki, 2018-06-17:
This is probably not a bug in the library but maybe someone here would help me understanding what's going on.

I have a small RPi monitoring server implementation that uses the pigpio library in order to control the Google Voice Kit button and LED light.

Repo is here: https://github.com/sznowicki/rpi-av-monitoring-server.

When the node process is starting, it starts the mjpeg (video) and darkice (audio) stream.

At the moment I init the pigpio, audio stream goes mad. [The full implementation is here](https://github.com/sznowicki/rpi-av-monitoring-server/tree/led), but in order to narrow the possible suspects i did a following test:

1. Start audio stream (darkice app, streams mp3 from alsa plug:micboost
2. Check if works -> works fine, no problems.
3. Open node project, import lib and init one device:
```js
const Gpio = require('pigpio').Gpio; // all good
const l = new Gpio(25, { mode: Gpio.OUTPUT }); 
```
4. At the moment `const l = new Gpio(25, { mode: Gpio.OUTPUT });` is called, darkice goes crazy throwing this warning repeatedly `AlsaDspSource :: Buffer overrun!`

---

I suspect that maybe, the library is abusing the connection on GPIO bus and makes "everything" too busy so the Google Voice Kit hat is not able to process audio any more?

Can it be the problem?

If not, I'd be glad for any suggestions how to deal with this issue.

fivdi, 2018-06-17:
Does it work if the following code is used?
```js
const pigpio = require('pigpio');
const Gpio = pigpio.Gpio;

// Call configureClock before creating Gpio objects
pigpio.configureClock(1, pigpio.CLOCK_PWM);

const l = new Gpio(25, { mode: Gpio.OUTPUT }); 
```

fivdi, 2018-06-17:
Here's a potentially related issue from the pigpio C library: https://github.com/joan2937/pigpio/issues/87

sznowicki, 2018-06-18:
Looks like a potential fix. I'll check that in the evening (CEST) and post my results here. Thanks for a quick feedback!

sznowicki, 2018-06-18:
It did fix the problem. Thanks a lot for pointing this out. If you find it useful, I'd be happy to edit the README by adding a Troubleshooting chapter with short description of my problem and links to this and C library ticket.

fivdi, 2018-06-18:
> It did fix the problem.

Excellent.

> Thanks a lot for pointing this out.

You're welcome.

> If you find it useful, I'd be happy to edit the README by adding a Troubleshooting chapter with short description of my problem and links to this and C library ticket.

This is a very good idea. Maybe in a file called `troubleshooting.md` in the doc directory. There could be a link to `troubleshooting.md` in the [Contents section](https://github.com/fivdi/pigpio#contents).

sznowicki, 2018-06-19:
I've created a pull request. Since I have only slight idea about the topic I describe, I tried to be as vague as possible.

https://github.com/fivdi/pigpio/pull/53

I also allowed myself to add an .gitignore entry. I hope this won't bother you. If not, can also remove it from the pull request.

fivdi, 2018-06-29:
@sznowicki thank you very much for the PR, it has just been merged.

closed 2018-06-29 by fivdi