fivdi/pigpio #56: Cannot get pwm demo running

LockonS, 2018-08-05:
Node.js version: `10.8.0`
pigpio version: `1.1.2`
pigpio C library already installed and started as root.
While trying to get demo running, it always output errors as below.

```
2018-08-05 21:17:28 initInitialise: Can't lock /var/run/pigpio.pid
/root/pi-gpio-trial/node_modules/pigpio/pigpio.js:16
    pigpio.gpioInitialise();
           ^

Error: pigpio error -1 in gpioInitialise
    at initializePigpio (/root/pi-gpio-trial/node_modules/pigpio/pigpio.js:16:12)
    at new Gpio (/root/pi-gpio-trial/node_modules/pigpio/pigpio.js:29:5)
    at Object.<anonymous> (/root/pi-gpio-trial/test.js:3:13)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:742:12)
    at startup (internal/bootstrap/node.js:266:19)
```
And the demo that I tried to run is 

```js
const Gpio = require('pigpio').Gpio;

const led = new Gpio(17, {mode: Gpio.OUTPUT});

let dutyCycle = 0;

setInterval(() => {
  led.pwmWrite(dutyCycle);

  dutyCycle += 5;
  if (dutyCycle > 255) {
    dutyCycle = 0;
  }
}, 20);
```

I tried to run the script as pi with `sudo` or directly run the script as root, but still cannot get it to run properly.

onokje, 2018-08-05:
That happens if the gpio is already in use by another process. Maybe you have it running twice? Quit other processes that may use the gpio and try again.

LockonS, 2018-08-05:
@onokje Thanks for the reply, but I am pretty sure that this node.js application is the only processess that use the gpio.

fivdi, 2018-08-05:
@LockonS please verify that the pigpiod daemon  is not running. If the pigpiod daemon is running the pigpio Node.js module can not be run.

LockonS, 2018-08-05:
@fivdi Thanks a lot and it works perfectly now. I was so stupid to miss the statement in readme. Maybe highlight this part in readme could save the some others in case I was not the only one to overlock that.