fivdi/pigpio #58: Tick utilities not implemented

bahamut657, 2018-09-01:
Hi
Is it possible to add implementation of these functions needed to bootup some devices? Actually,there are no functions to retrieve these datas and process.hrtime() is a bit different
`get_current_tick`
`tickDiff`

fivdi, 2018-09-02:
The pigpio Node.js module could be extended to provide access to the pigpio C library [gpioTick](http://abyz.me.uk/rpi/pigpio/cif.html#gpioTick) function which returns the current tick. `gpioTick` is the pigpio C library equivalent of the pigpio Python `get_current_tick` function.

Unlike the the pigpio Python library the pigpio C library does not have a `tickDiff` function. However, it would be easy to add a `tickDiff` function implemented in pure JavaScript to the pigpio Node.js module. To see how to determine the difference between two ticks in JavaScript see the documentation for the [alert event](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#event-alert).

bahamut657, 2018-09-07:
This can be closed merging the Waves branch to Master

fivdi, 2018-09-07:
@bahamut657 thank you for the information.

erikma, 2018-10-21:
Looking for the same functionality and see it in your Waves fork, @bahamut657 . When is that branch coming to master? If not soon I may need to create a PR with just that functionality to resolve this enhancement request.

bahamut657, 2018-10-22:
@erikma I'm working on it. I'm thinking it will available this weekend.
Regards
MaDPapo

erikma, 2018-10-29:
@bahamut657 I see your PR for Waves in #64, my PR #65 is mostly orthogonal except for exposing gpioTick() from pigpio.cc. I placed the getTick() and tickDiff() methods at the module level in my implementation, where yours adds to the GPIO class. Both are valid approaches and IMO should just be merged together to expose both methods for ease of use.

erikma, 2018-11-18:
@fivdi  any chance of completing #65 to help resolve this issue?

fivdi, 2018-11-25:
@bahamut657 @erikma Thank you for suggesting this enhancement and the pull request with the corresponding implementation. pigpio@1.2.0 which has been published on npm contains the enhancement.

closed 2018-11-25 by fivdi

erikma, 2018-11-25:
@fivdi Thanks for accepting the PR and publishing! Corresponding TypeScript definition updates in PR at https://github.com/DefinitelyTyped/DefinitelyTyped/pull/30811 . pigpio 1.2.0 validated and placed into production in my system. pigpio-dht PR using 1.2.0 to fix repeated readings: https://github.com/depuits/pigpio-dht/pull/1