fivdi/pigpio #59: pigpio in Docker container running raspberrypi3-node

mdbeauche, 2018-09-06:
I'm having issues getting pigpiod to run in a default Docker container running raspberrypi3-node.

Dockerfile:

```
FROM resin/raspberrypi3-node

RUN apt-get update && apt-get install -yq --no-install-recommends pigpio

RUN npm install pigpio
```

From running docker (`docker run -it name /bin/bash` and then `sudo pigpiod` at shell as root):

```
2018-09-06 19:55:38 initCheckPermitted: 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

root@1bd67dc5c6e4:/# Can't initialise pigpio library
```

I've tried changing docker run permissions (`--privileged` and `--cap-add=ALL`) and I get the same result.

fivdi, 2018-09-07:
@mdbeauche This repository contains the code for a Node.js wrapper for the [pigpio C library](https://github.com/joan2937/pigpio). The pigpiod daemon is part of the pigpio C library, not part of the Node.js wrapper for the pigpio C library. Please open a new issue at [pigpio C library](https://github.com/joan2937/pigpio/issues) for help.

closed 2018-09-07 by fivdi

grevelle, 2025-07-31:
## ✅ **SOLVED: Working pigpio + Docker Solution (2025)**

After extensive research and testing, I've found the **minimal working configuration** for running the pigpio npm package in Docker containers. The key insight is that **direct hardware access** works better than daemon-based approaches.

### **Complete Working Solution**

#### **docker-compose.yml**
```yaml
services:
  app:
    privileged: true
    devices:
      - "/dev/gpiomem:/dev/gpiomem"
    volumes:
      - /sys/class/gpio:/sys/class/gpio
    # ... your other config
```

#### **Application Code**
```javascript
import pigpio from 'pigpio';

const initializePigpio = () => {
  try {
    pigpio.initialize(); // Direct hardware access
    console.log('✅ pigpio initialized successfully');
    // Initialize GPIO-dependent components here
  } catch (err) {
    console.log(`❌ Failed to initialize pigpio: ${err.message}`);
    console.log('Continuing without GPIO functionality...');
  }
};

initializePigpio();
```

### **Why This Works**

1. **No pigpiod daemon needed** - The pigpio npm package uses the C library directly
2. **Minimal device mapping** - Only `/dev/gpiomem` is required for GPIO memory access  
3. **Essential volume mount** - `/sys/class/gpio` provides the GPIO sysfs interface
4. **Privileged mode** - Required for hardware access permissions

### **What NOT to Do**

❌ **Don't try to connect to pigpiod** - The npm package doesn't support remote connections  
❌ **Don't use environment variables** like `PIGPIO_ADDR` - They don't work with the npm package  
❌ **Don't map all devices** - Most are unnecessary (`/dev/mem`, `/dev/gpiochip*`, etc.)

### **Verified Results**

**Tested on:**
- Raspberry Pi CM4
- Docker Compose v2  
- pigpio npm package v3.3.1
- Raspberry Pi OS (64-bit)

**Success indicators:**
```
✅ pigpio initialized successfully
GPIO is ready - initializing GPIO-dependent components...
```

### **Performance**

This minimal approach provides:
- Clean initialization without errors
- Full GPIO functionality (PWM, interrupts, etc.)
- No overhead from daemon management
- Universal compatibility (Docker + native environments)

### **Troubleshooting**

If you still get permission errors:
1. Verify `privileged: true` is set
2. Check that `/dev/gpiomem` exists on host: `ls -la /dev/gpiomem`
3. Ensure you're running on actual Raspberry Pi hardware

This solution eliminates all the complexity around daemon management and provides reliable GPIO access in containers.