fivdi/pigpio #62: Error using pigpio for long time

bahamut657, 2018-09-17:
Hi @fivdi 
I've build a services based on PIGPIO library to manage various sensor connected to my RPI3B+
After 4-6 hours of inactivity the process using PIGPIO library return this error

`pthSocketThreadHandler: ext too large 1919251247(65536), sock=17`

Any ideas about the root cause and the possible solution?
Regards
Jacopo

bahamut657, 2018-09-24:
It is something related to interrupts on a GPIO where an INPUT sensor is connected.
I try to isolate the case and post the code to reproduce as soon as possible

fivdi, 2018-10-12:
@bahamut657 any news on this one?

bahamut657, 2018-10-15:
The code that, running a 433Receiver (SRX882 - that is sending much noise/signal on line), reproduces the message is 
```
pigpio.enableInterrupt(pigpio.EITHER_EDGE);
pigpio.on("interrupt",(tick,interrupt)=>{
    //HERE I CALL MY CALLBACK THAT IS CALLED MUCH MUCH TIME
})
```
I obtain it after 8hour the task is running
I don't know if it is related to the NodeJS implementation or a real warning derivated by the sensor connected

fivdi, 2018-10-15:
I'm afraid that's not much to go on. I need a complete sample that can be used to reproduce the error.

The error itself comes from [here in the pigpio C library](https://github.com/joan2937/pigpio/blob/V67/pigpio.c#L6935-L6936) and has something to do with sockets.

fivdi, 2018-10-15:
Do you know approximately how many interrupts occur per second on your system?

bahamut657, 2018-10-16:
Yes the error is regarding socket used to communicate with PIGPIO daemon I think.
How to give you a result on how many interrupt per second? I introduce a counter variable in my code to count the "interrupt event" occurence.Or have you got some tools to measure it?

Let me know, and I will test it this evening

I think that this kind of sensor send too much interrupt (I don't know why, I fot 3 units per model here and all "make noise on line")

fivdi, 2018-10-16:
> Yes the error is regarding socket used to communicate with PIGPIO daemon I think.

When using the pigpio Node.js module the pigpio daemon does not (and can not) run.

> How to give you a result on how many interrupt per second? ?Do I have to introduce a counter variable in my code to count the "interrupt event" occurences?Or have you got some tools to measure it?

A counter should work. Something like this:
```js
let second = 0,
let interruptCount = 0;
let totalInterruptCount = 0;

setInterval(() => {
    second += 1;
    totalInterruptCount += interruptCount;
    console.log(second, totalInterruptCount, interruptCount);
    interruptCount = 0;
}, 1000);

pigpio.on("interrupt",(tick,interrupt)=>{
    //HERE I CALL MY CALLBACK THAT IS CALLED MUCH MUCH TIME
    interruptCount += 1;
})
```

bahamut657, 2018-10-16:
I tried to measure interrupt count. I can say that:

1. While the sensor is in IDLE (no 433 transmitter receiving) I got, per second, 350 interrupts ca. in 80% times, 15% 700interrupts, 5% 900interrupts

2. While receiveing (activating 433 transmitter on other server) I got from 950 to 1100 interrupts per second 100% of the time

Is it an high value?
Let me know I to investigate further more

Regards
MaDPapo

fivdi, 2018-10-16:
I'm afraid I'm not familiar with the SRX882 but 350 to 900 interrupts per second when idle sounds unusual to me. To be honest I wouldn't expect any interrupts when the SRX882 is idle. But as already mentioned I'm not familiar with the SRX882.

fivdi, 2019-06-08:
I'm going to go ahead and close this issue as it doesn't appear to be an issue with the pigpio JavaScript moudule.

closed 2019-06-08 by fivdi