fivdi/pigpio #63: Usage of enableInterrupts() question

ezehurt, 2018-10-11:
My objetive is to read RISING_EDGE pulses from hardware, but I'm getting an error trying to enabling interrupts
Here is the declaration of the pin
`const pulsesPin= new Gpio(6, {mode: Gpio.INPUT,pullUpDown: Gpio.PUD_OFF,edge: Gpio.RISING_EDGE, interrupts:true});`

Here I have a snippet I used de method disableInterrupt
`
	socket.on('stopPulsesCounter', function(socket){
		pulsesPin.disableInterrupt();	
		io.sockets.emit('pulseRecieved', {pulses : quanty, last:true }),
		clearInterval(killId);
		});
`
In other part of the code I want to enable interrupts?

The especification says
enableInterrupt(edge[, timeout])
edge - RISING_EDGE, FALLING_EDGE, or EITHER_EDGE
timeout - interrupt timeout in milliseconds (optional, defaults to 0 meaning no timeout)

I wrote 
`pulsesPin.enableInterrupt()`
But im getting this error

  pigpio.gpioSetISRFunc(this.gpio, +edge, +timeout, handler);
           ^

Error: EINVAL, Invalid argument
    at Gpio.enableInterrupt (/home/pi/rimagro-dispenser-connector/node_modules/pigpio/pigpio.js:135:12)
    at Socket.<anonymous> (/home/pi/rimagro-dispenser-connector/index.js:47:13)
    at Socket.emit (events.js:182:13)
    at /home/pi/rimagro-dispenser-connector/node_modules/socket.io/lib/socket.js:528:12

fivdi, 2018-10-12:
The [enableInterrupt](https://github.com/fivdi/pigpio/blob/master/doc/gpio.md#enableinterruptedge-timeout) method accepts two arguments, `edge` and `timeout`. `timeout` is optional but `edge` isn't and must be specified when `enableInterrupt` is invoked.

Try with:
```js
 pulsesPin.enableInterrupt(Gpio.RISING_EDGE)
```
rather than:
```js
 pulsesPin.enableInterrupt()
```
I'm sorry that the error message "Error: EINVAL, Invalid argument" isn't as helpful as it could be.

fivdi, 2018-10-14:
@ezehurt I'm going to go ahead and close this issue as it's not a pigpio issue. If it does turn out to be a pigpio problem the issue can be reopened.

closed 2018-10-14 by fivdi