fivdi/pigpio #68: .servoWrite hang after completion

PierreTrp, 2018-12-21:
I'm controlling a servo like this:

```js
const TRAP = new Servo(5, {mode: Servo.OUTPUT});

function openTrap() {
    TRAP.servoWrite(1400);
}

function closeTrap() {
    TRAP.servoWrite(2200);
}
```
If i call `openTrap()` / `closeTrap()`, the servo will rotate to the desired position but then i hear a noise, like it was not finished rotating, but it's not (and i tried lower/higher value between the correct range allowed for my servos). 

When i halt my node application, the noise stop, obviously.

Does there is a command where i can stop any write on the servo after using it ? (i need to be able to re-use it later thought)

fivdi, 2018-12-21:
pigpio doesn't have a class called `Servo` so I'm not 100% sure how it's `servoWrite` method is implemented but I think I can guess.

In the sample code above I wouldn't expect `TRAP.servoWrite(1400)` to result in noise coming from the servo. Is this really the case?

Calling `TRAP.servoWrite(2200)` is probably the call that results in the noise, correct?

Each servo is a little different and they need to be calibrated. In general, they should function without issue with servo pulse widths from 1000 microseconds to 2000 microseconds. The further you go outside this range the more likely you are to hear noises coming from the servo. 2200 is a fair bit outside this range. It's not a good idea to run the servo with pulses that result in this noise as it may break the servo.

Are you using an inexpensive hobby servo? If so, don't be too surprised if it doesn't do exactly what you expect.

> Does there is a command where i can stop any write on the servo after using it ? (i need to be able to re-use it later thought)

To achieve this invoke `digitalWrite(0)` on the appropriate `Gpio` object.

fivdi, 2018-12-27:
@PierreTrp Were you able to make any progress here? I don't think it's a pigpio issue.

PierreTrp, 2018-12-27:
`Servo` was just `const Servo = require('pigpio').Gpio;`

I indeed use cheap servos, but only one of the 4 make this noise so maybe it's just the servo.

I will try `digitalWrite(0)` asap thanks for the help

closed 2018-12-27 by PierreTrp