fivdi/pigpio #69: How to set servo using angle degree?

simar88, 2019-01-05:
Hi everyone,
i would like to know how could be possible to set the servo angle using degree instead of dutycicle.
Do you have some ideas?
Thanks in advance,
Simone

fivdi, 2019-01-06:
Yes, this is possible.

Let's say you have a servo and the pulse width for moving to an angle of 0° is 1000 microseconds and the pulse width for moving to an angles of 180° is 2000 microseconds.

If the servo is connected to GPIO10, the following program will continuously move the servo from 0° to 180° and back again.

```js
const Gpio = require('pigpio').Gpio;

const servo = new Gpio(10, {mode: Gpio.OUTPUT});

const PULSE_WIDTH_0 = 1000; // Pulse width in microseconds for 0 degrees
const PULSE_WIDTH_180 = 2000; // Pulse width in microseconds for 180 degrees

function moveServoTo(angle) {
  let microSeconds = PULSE_WIDTH_0 + (angle / 180) * (PULSE_WIDTH_180 - PULSE_WIDTH_0);
  servo.servoWrite(Math.round(microSeconds));
}

let angle = 0;
let increment = 1;

setInterval(() => {
  moveServoTo(angle);

  angle += increment;
  if (angle >= 180) {
    increment = -1;
  } else if (angle <= 0) {
    increment = 1;
  }
}, 25);
```
The important line which converts degrees to microseconds is:
```js
  let microSeconds = PULSE_WIDTH_0 + (angle / 180) * (PULSE_WIDTH_180 - PULSE_WIDTH_0);
```

fivdi, 2019-01-08:
I'll go ahead and close this issue as I think the question has been answered.

closed 2019-01-08 by fivdi