fivdi/pigpio #70: node-gyp build failure (This version of node/NAN/v8 requires a C++11 compiler)

dbush3721, 2019-01-30:
pi@helmsdeep ~/devel/drawbot $ node -v; npm -v; pigpiod -v
v8.15.0
6.4.1
68

pi@helmsdeep ~/devel/drawbot $ npm i pigpio

> pigpio@1.2.1 install /home/pi/devel/drawbot/node_modules/pigpio
> node-gyp rebuild

make: Entering directory '/home/pi/devel/drawbot/node_modules/pigpio/build'
  CXX(target) Release/obj.target/pigpio/src/pigpio.o
In file included from ../src/pigpio.cc:3:0:
../../nan/nan.h:49:3: error: #error This version of node/NAN/v8 requires a C++11 compiler
In file included from /home/pi/.node-gyp/8.15.0/include/node/node.h:63:0,
                 from ../../nan/nan.h:53,
                 from ../src/pigpio.cc:3:

[2019-01-30T02_03_16_073Z-debug.log](https://github.com/fivdi/pigpio/files/2810577/2019-01-30T02_03_16_073Z-debug.log)

fivdi, 2019-02-02:
The pigpio module depends on [nan](https://github.com/nodejs/nan) which in turn requires a [C++ 11 compiler](https://github.com/nodejs/nan/blob/v2.12.1/nan.h#L49).

If you are using a fairly recent version of Raspbian as the operating system on your Raspberry Pi then everything should work correctly. To the best of my knowledge gcc 4.8 or higher is needed to compile pigpio and nan when installing pigpio. Check which version of gcc is being used with the command `gcc --version`. Is the version number 4.8 or higher?

Raspbian Wheezy (which is now quite old) comes with gcc 4.6. This [article](https://github.com/fivdi/onoff/wiki/Node.js-v4-and-native-addons) explains how to install gcc 4.8 on Raspbian Wheezy.

dan3721, 2019-02-02:
Thanks, I was able to upgrade gcc and all is well now.