fivdi/pigpio #72: sudo node doorlock.js help don't know what to do urgent

jjhamer123, 2019-02-21:
pi@raspberrypi:~/Smartphone-Doorlock $ sudo node doorlock.js
/home/pi/Smartphone-Doorlock/node_modules/pigpio/node_modules/bindings/bindings.js:91
  throw err
  ^

Error: Could not locate the bindings file. Tried:
 → /home/pi/Smartphone-Doorlock/node_modules/pigpio/build/pigpio.node
 → /home/pi/Smartphone-Doorlock/node_modules/pigpio/build/Debug/pigpio.node
 → /home/pi/Smartphone-Doorlock/node_modules/pigpio/build/Release/pigpio.node
 → /home/pi/Smartphone-Doorlock/node_modules/pigpio/out/Debug/pigpio.node
 → /home/pi/Smartphone-Doorlock/node_modules/pigpio/Debug/pigpio.node
 → /home/pi/Smartphone-Doorlock/node_modules/pigpio/out/Release/pigpio.node
 → /home/pi/Smartphone-Doorlock/node_modules/pigpio/Release/pigpio.node
 → /home/pi/Smartphone-Doorlock/node_modules/pigpio/build/default/pigpio.node
 → /home/pi/Smartphone-Doorlock/node_modules/pigpio/compiled/10.15.1/linux/arm/pigpio.node
    at bindings (/home/pi/Smartphone-Doorlock/node_modules/pigpio/node_modules/bindings/bindings.js:88:9)
    at Object.<anonymous> (/home/pi/Smartphone-Doorlock/node_modules/pigpio/pigpio.js:5:31)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at Object.<anonymous> (/home/pi/Smartphone-Doorlock/doorlock.js:47:12)

fivdi, 2019-02-21:
As mentioned in the error message, the bindings file can't be found. The bindings file is typically located at `node_modules/pigpio/build/Release/pigpio.node`. Does this file exist on your Pi? If not, I would imagine that pigpio wasn't installed correctly for some reason.

Were the [installation instructions](https://github.com/fivdi/pigpio#installation) followed?
What is the complete output of the command `npm install pigpio`?

fivdi, 2019-02-23:
@jjhamer123 I'll go ahead and close this as it's most likely an installation issue. If it turns out to be a problem with pigpio feel free to reopen the issue.

closed 2019-02-23 by fivdi