fivdi/pigpio #74: gpioSetPWMfrequency: bad gpio (32)

ronaldroe, 2019-05-06:
I'm getting the error in the title every time I try to do anything with PWM. Digital writes seem to work without issue, but as soon as anything tries to write PWM, the error pops up.

I'm trying to reimplement a motor controller's Python library in Node. The LED pulse example seems to work without issue. I'm sure I'm just missing something. 

I'll post the code for the class and constructor below. It's a long file, so let me know if what's below isn't enough, and I can post the whole thing.

```
'use strict';
const Gpio = require('pigpio').Gpio;

class Motor{
  
  /* 
    Class to handle interaction with the motor pins
    Supports redefinition of "forward" and "backward" depending on how motors are connected
    Use the supplied Motorshieldtest module to test the correct configuration for your project.
    
    Arguments:
      motor = string motor pin label (i.e. "MOTOR1","MOTOR2","MOTOR3","MOTOR4") identifying the pins to which
              the motor is connected.
      config = number defining which pins control "forward" and "backward" movement.
  */
  
  constructor(motor, config){
    
    this.motorpins = {
      MOTOR4: {
        config: {
          1: {e:32,f:24,r:26},
          2: {e:32,f:26,r:24}
        },
        arrow:1
      },
      MOTOR3: {
        config: {
          1: {e:19,f:21,r:23},
          2: {e:19,f:23,r:21}
        }, 
        arrow:2
      },
      MOTOR2: {
        config: {
          1: {e:22,f:16,r:18},
          2: {e:22,f:18,r:16}
        }, 
        arrow:3
      },
      MOTOR1: {
        config: {
          1: {e:11,f:15,r:13},
          2: {e:11,f:13,r:15}
        },
        arrow:4
      }
    };
    
    this.testMode = false;
    
    this.arrow = new Arrow(this.motorpins[motor].arrow);
    
    this.pins = this.motorpins[motor].config[config];
    
    this.motorE = new Gpio(this.pins.e, {mode: Gpio.OUTPUT});
    this.motorF = new Gpio(this.pins.f, {mode: Gpio.OUTPUT});
    this.motorR = new Gpio(this.pins.r, {mode: Gpio.OUTPUT});
    
    this.motorE.pwmFrequency(50);
    
    this.motorF.digitalWrite(0);
    this.motorR.digitalWrite(0);
    
  }
// The rest of the class would follow. 
```
The error is thrown at `this.motorE.pwmFrequency(50);`. I've tried commenting that line out, and it just throws the error on the next PWM call further down.

fivdi, 2019-05-06:
pigpio uses GPIO numbers rather than pin numbers. Pin 32 is GPIO12 so the number to use is 12, not 32.

ronaldroe, 2019-05-06:
Oh, wow. I really should have noticed that.

I changed them all, and now it works like a charm. Thank you.

closed 2019-05-06 by ronaldroe