fivdi/pigpio #75: Error when calling pwmWrite()

ralphhughes, 2019-05-14:
I'm trying to write a motor controller for a servomotor with optical encoder and I'm getting this error sporadically. I can't pin down what's causing it as I can't reproduce it every time. Any idea where I should start to look?

```
pigpio/pigpio.js:98
    pigpio.gpioPWM(this.gpio, +dutyCycle);
           ^

Error: EINVAL, Invalid argument
    at Gpio.pwmWrite
```

fivdi, 2019-05-14:
dutyCycle must be an unsigned 32-bit integer. When calling pwmWrite be sure to pass an unsigned 32-bit integer rather than a number. A number can be converted to an unsigned 32-bit integer with the right-shift operator. Use something like this:
```js
gpio.pwmWrite(dutyCycle >> 0);
```

ralphhughes, 2019-05-14:
That appears to have fixed it so far, many thanks for your accurate and timely response.

fivdi, 2019-05-14:
That's good news. Thanks for the feedback.

closed 2019-05-14 by fivdi