fivdi/pigpio #76: pigpio unitialised before digitalWrite()

dlingo, 2019-06-03:
Hi I am having trouble ending process gracefully and resetting gpio to a LOW state on exit.

Code:

```javascript
import {exec} from "child_process";
import * as pigpio from "pigpio";
import {Gpio} from "pigpio";

const hwRelays = [5,6,13,16,19,20,21,26]

console.log("Initializing pigpio")
pigpio.initialize(); 
// pigpio initialized here...

exec('sleep 3', () => {
  console.log("3 second Delay");
});

process.on('SIGINT', () => {
  relay1.digitalWrite(0);
  relay2.digitalWrite(0);
    pigpio.terminate(); 
  // pigpio terminated here...
  console.log("Terminating  pigpio");
});

const relay1 = new Gpio(hwRelays[0])
const relay2 = new Gpio(hwRelays[1])

```
Error:
```
2019-06-03 22:22:12 gpioWrite: pigpio uninitialised, call gpioInitialise()
/home/pi/client/node_modules/pigpio/pigpio.js:88
    pigpio.gpioWrite(this.gpio, +level);
           ^

Error: pigpio error -31 in gpioWrite
    at Gpio.digitalWrite (/home/pi/client/node_modules/pigpio/pigpio.js:88:12)
    at Timeout.setTimeout [as _onTimeout] (/home/pi/client/dist/index.js:40:16)
    at listOnTimeout (internal/timers.js:535:17)
    at processTimers (internal/timers.js:479:7)
```

fivdi, 2019-06-05:
If ctrl+c is hit when running the following program with Node.js v12.4.0 and pigpio v1.2.2
```js
const exec = require("child_process").exec;
const pigpio = require("pigpio");
const Gpio = pigpio.Gpio;

const hwRelays = [17,18]

console.log("Initializing pigpio")
pigpio.initialize(); 
// pigpio initialized here...

exec('sleep 3', () => {
  console.log("3 second Delay");
});

process.on('SIGINT', () => {
  relay1.digitalWrite(0);
  relay2.digitalWrite(0);
    pigpio.terminate(); 
  // pigpio terminated here...
  console.log("Terminating  pigpio");
});

const relay1 = new Gpio(hwRelays[0], {mode: Gpio.OUTPUT})
const relay2 = new Gpio(hwRelays[1], {mode: Gpio.OUTPUT})
```
the output is
```
Initializing pigpio
^CTerminating  pigpio
3 second Delay
```
Everything works as expected and no error is thrown.

fivdi, 2019-06-06:
The following stacktrace is shown above
```
2019-06-03 22:22:12 gpioWrite: pigpio uninitialised, call gpioInitialise()
/home/pi/client/node_modules/pigpio/pigpio.js:88
    pigpio.gpioWrite(this.gpio, +level);
           ^

Error: pigpio error -31 in gpioWrite
    at Gpio.digitalWrite (/home/pi/client/node_modules/pigpio/pigpio.js:88:12)
    at Timeout.setTimeout [as _onTimeout] (/home/pi/client/dist/index.js:40:16)
    at listOnTimeout (internal/timers.js:535:17)
    at processTimers (internal/timers.js:479:7)
```
This indicates that `Gpio.digitalWrite` was called on line 40 of `index.js`. However, the code posted above doesn't contains 40 lines. The stacktrace is not from the code posted. It's from something else.

fivdi, 2019-06-08:
@dlingo any news on this one?

dlingo, 2019-06-10:
@fivdi I updated my node install on the pi to V12.4 and pigpio to V1.2.2 and the problem was resolved. Yarn installed an earlier version because it could not find a type file for V1.2.2. Thanks for the help

closed 2019-06-10 by dlingo

fivdi, 2019-06-10:
@dlingo good to hear that it was possible to resolve the issue and thank you for the feedback.