fivdi/pigpio #77: How do I release my pins?

guotingchaopr, 2019-06-12:
now, i release these pins 
```js
clearServePin() {
    new Gpio(this.STEER_PIN, {mode: Gpio.PUD_OFF});
  }
```

Always feeling embarrassed, so what should be the right approach?

fivdi, 2019-06-12:
I'm not 100% sure what "releasing a pin" means here but I think it means "how do I get a pin back into the mode it was in before my program started?".

If this is the case, then the pigpio Gpio methods `mode` and `pullUpDown`can typically be used to return the pin to the mode it was in before a program started.

For example, when a Raspberry Pi is booted, GPIO17 is configured as an input and it's pull-down resistor is activated. Lets say you have an LED connected to GPIO17 and a program that configures GPIO17 as an output and turns the LED on. The following program uses the `mode` and `pullUpDown` methods to return GPIO17 to the appropriate mode before terminating:
```js
const Gpio = require('pigpio').Gpio;

// Create a Gpio object for the LED on GPIO17.
const led = new Gpio(17, {mode: Gpio.OUTPUT});

// Turn LED on.
led.digitalWrite(1);

// Wait 2 seconds before terminating the program.
setTimeout(() => {
  // Reconfigure GPIO17 as an INPUT
  led.mode(Gpio.INPUT);

  // Activate the pull-down resistor on GPIO17.
  led.pullUpDown(Gpio.PUD_DOWN);
}, 2000);
```

guotingchaopr, 2019-06-13:
@fivdi Tks, You solved my confusion.

closed 2019-06-13 by guotingchaopr